NSPManager/NSPanelManager #173: [Bug] URL manager not showed

mainmind83, 2024-09-01:
Missing administrator URL and port with new firmware (not migrated)

1. Update firmware to 0.1.150
2. Visit Website of panel
3. Manager address show  -- UNKNOW TEMPLATE -- 
4. Manager port (blank)

[image]

**Environment info:**
 - Home Assistant 2024.8.3
 - Chrome
 - NSPanelManager version 1.0.0
 - NSPanel Firmware version 0.1.150

tpanajott, 2024-09-01:
Yhea, ignore that. This happens because it has updated the firmware and not littlefs. Try updating the firmware again and make sure you have updated your addon/container.

mainmind83, 2024-09-01:
Mmm I have tried to set the value and save the values, now the panels appear offline on manager...

tpanajott, 2024-09-01:
Did you set the manager address and port in the manager?

mainmind83, 2024-09-01:
Yes, of course... i updated from previuos version, HA show wizard en updating too...

tpanajott, 2024-09-01:
Okay, just double checking. Do you have logging enabled on the panel?

tpanajott, 2024-09-01:
You should be able to see logs from the panel in the manager given you have logging turned on on the panel. Otherwise, reboot the panel and in the web interface you showed in the picture above set the log level to debug.

mainmind83, 2024-09-01:
[image]

[image]

Not showed protocol versión... not showed log...

mainmind83, 2024-09-01:
[image]

There is some problem when retrieving the data, I have set everything again (including MQTT password) which seems not to be correct when reloading the interface text box
NOTE: I think I remember that I reported a similar bug in the previous version... I'm going to look

mainmind83, 2024-09-01:
After re-entering the MQTT password, the configuration has been saved, I have updated the firmware again (second time) and now the v0.1.150 website does look like the new one (different options and blue background), I am updating TFT...

[image]

tpanajott, 2024-09-01:
Nice progress. Denna like you are getting somewhere!

mainmind83, 2024-09-01:
I had deleted browser cache, private mode... in testing.

It should not be necessary to send the firmware twice to the device for it to update the web interface completely...

tpanajott, 2024-09-01:
That's a known bug we had previously that's been fixed. It shouldn't happen again.

mainmind83, 2024-09-01:
Fixed, if anyone else has this problem:

1.- Enter the panels website and enter the data again ([including the MQTT server password information](https://github.com/NSPManager/NSPanelManager/issues/82))
2.- Restart NSPanelManager
3.- Send firmware update again
4.- Update TFT

closed 2024-09-01 by mainmind83