NSPManager/NSPanelManager #174: Show outside temperature on non-weather screensaver

tpanajott, 2024-09-01:
**Is your feature request related to a problem? Please describe.**
It would be nice to be able to display outside temperature on the non-weather screensaver.

**Describe the solution you'd like**
Add option to display outside temperature from weather data or sensor in the upper left corner on the "only time" screensaver. Just the same as the curren room temperature in upper right but perhaps in upper left? Make it optional and also make showing room temperature optional.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

cablesandcoffee, 2024-09-02:
Okay, understand. So for the tft file I will just copy the outside temperature entities from weather screensaver and add them to the only time screensaver page.

tpanajott, 2024-09-02:
Hm, well the outside temperature is really big in the weather screensaver. I was thinking more the size of the inside room temperature in the upper right. Then we can easily hide them by setting the visibility from firmware.

cablesandcoffee, 2025-03-09:
Updated the hmi files in devel branch. I might have to update the defines file as well or whatever its called. Will check that out tomorrow.

cablesandcoffee, 2025-03-10:
HMI files and GUI_data file are now updated.

tpanajott, 2025-03-11:
This should not work as expected, if so please close issue.

cablesandcoffee, 2025-03-15:
Nicely done, closing! 💯

closed 2025-03-15 by cablesandcoffee