NSPManager/NSPanelManager #179: Latin Extended font implementation

oleyek, 2024-09-04:
There is a problem with names of lights, switches or scenes in other languages than Western Languages. Is possible to add support for Latin Extended characters for Central/East EU languages?

tpanajott, 2024-09-04:
This is mainly an issue with the TFT file and Nextion I believe. Do @cablesandcoffee have any thoughts on this?

cablesandcoffee, 2024-09-06:
Hi oleyek! It is definitely possible. Can you give me an example of a few characters that you want to use? It is just so I search in the right place. I have to refresh my mind on all the different character sets that exists and how I did to add characters when we did this the first time around. :)

oleyek, 2024-09-07:
Hi @cablesandcoffee I glad to hear that. :)

I was mainly referring to Polish diacritical marks: “ą, ć, ę, ł, ń, ó, ś, ź, ż, Ą, Ć, Ę, Ł, Ń, Ó, Ś, Ź, Ż” Now there are blank spaces instead of these characters. Maybe except letter "Ó and ó" - because this character exist in some western languages.

cablesandcoffee, 2024-09-09:
Okay, nice. I was waiting for someone to ask for more letters. It's a good sign since it means we are reaching more people! :) When figured out this first time around we just added Swedish weird letters but it's time to expand that now. Here's my thoughts:

Until now weäve only had the first two rows. C0 and C1
C0 Control and Basic Latin:
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~

C1 Control and Latin 1 Supplement:
¡¢£¤¥¦§¨©ª«¬­®¯°±²³´µ¶·¸¹º»¼½¾¿ÀÁÂÃÄÅÆÇÈÉÊËÌÍÎÏÐÑÒÓÔÕÖ×ØÙÚÛÜÝÞßàáâãäåæçèéêëìíîïðñòóôõö÷øùúûüýþÿ

I think we would cover a lot more by adding the two below, Latin extended A and B. You can find your letters in there right?

Latin Extended-A:
ĀāĂăĄąĆćĈĉĊċČčĎďĐđĒēĔĕĖėĘęĚěĜĝĞğĠġĢģĤĥĦħĨĩĪīĬĭĮįİıĲĳĴĵĶķĸĹĺĻļĽľĿŀŁłŃńŅņŇňŉŊŋŌōŎŏŐőŒœŔŕŖŗŘřŚśŜŝŞşŠšŢţŤťŦŧŨũŪūŬŭŮůŰűŲųŴŵŶŷŸŹźŻżŽžſ

Latin Extended-B:
ƀƁƂƃƄƅƆƇƈƉƊƋƌƍƎƏƐƑƒƓƔƕƖƗƘƙƚƛƜƝƞƟƠơƢƣƤƥƦƧƨƩƪƫƬƭƮƯưƱƲƳƴƵƶƷƸƹƺƻƼƽƾƿǀǁǂǃǄǅǆǇǈǉǊǋǌǍǎǏǐǑǒǓǔǕǖǗǘǙǚǛǜǝǞǟǠǡǢǣǤǥǦǧǨǩǪǫǬǭǮǯǰǱǲǳǴǵǶǷǸǹǺǻǼǽǾǿȀȁȂȃȄȅȆȇȈȉȊȋȌȍȎȏȐȑȒȓȔȕȖȗȘșȚțȜȝȞȟȠȡȢȣȤȥȦȧȨȩȪȫȬȭȮȯȰȱȲȳȴȵȶȷȸȹȺȻȼȽȾȿɀɁɂɃɄɅɆɇɈɉɊɋɌɍɎɏ

I suggest we start with that and wait for someone else to ask for even more letters? The more letters we add the bigger the TFT file but this is so far a small change.

oleyek, 2024-09-15:
Latin Extended-A covers all EU languages and even more. I think that A and B covers all characters of languages using latin letters. Central EU languages (included Polish) are in Extended-A. 

Only users of non-Latin languages (such as Cyrillic, Hebrew, Arabic, Chinese, etc.) can ask for more characters.

cablesandcoffee, 2024-09-17:
Okay, so the font we use only supported Latin extended A. Therefor I started with just adding latin A. If someone asks for Latin B we'll change font to something that support it. 

It's pretty easy to fix it now that I refreshed my mind.

So @oleyek, it would be great if you would volunteer to test. I've uploaded the new files to the beta branch.
To test, download them and use the upload function in the manager webui and choose upload tft file. After uploading tft file try uploadin to panel by pressing upload GUI on panel.

Eu file:
https://github.com/NSPManager/NSPanelManager/blob/beta/docker/web/nspanelmanager/gui.tft

US file:
https://github.com/NSPManager/NSPanelManager/blob/beta/docker/web/nspanelmanager/gui_us.tft

oleyek, 2024-09-18:
Hi @cablesandcoffee I've uploaded as you asked and now works perfectly, 👍. Polish characters are displayed correctly. Thank you for the fix 🤗

I have only one observation, regarding the design of the letter "J", it looks weird. It looks that way earlier too, but I think that may be the font related issue.
[image]

cablesandcoffee, 2024-11-21:
@oleyek super late answer but I'm closing this now. The J looks like that in the font so I think we will just live with that. 😊