NSPManager/NSPanelManager #180: Bug Report: Failed to connect to MQTTManager

Squal78, 2024-09-08:
**Description*
NspanelManager is failing to establish a connection with MQTTManager. The console displays a recurring error message indicating that the connection attempt has failed and that the system is continuously retrying to connect.

**Steps to Reproduce*

Start NspanelManager
Observe the console output

**Expected Behavior*
NspanelManager should successfully connect to MQTTManager without any error messages.

**Actual Behavior*
The console repeatedly displays the following error message:
CopyFailed to connect to MQTTManager. Retrying...
This message appears to loop, indicating that the connection attempts are consistently failing.

**Logs*
I get this issue on docker logs 
```
*** Operational MODE: preforking ***
Did not find a running MQTTManager, starting MQTTManager...
/MQTTManager/build/nspm_mqttmanager: error while loading shared libraries: /MQTTManager/build/libMQTTManager_WebHelper.so: unexpected reloc type 0x03
```

**Environment*
OS: Hassio + Supervisor + nspanelmanager (1.0.1)

**Environment info (please complete the following information):**
 - Home Assistant
 - NSPanelManager version [Web UI lower left corner]

tpanajott, 2024-09-08:
Hm, strange. What hardware are you running on? Specifically processor architecture.

tpanajott, 2024-09-09:
Hm, could you verify what arch of the container hass downloaded? To to the system cli and run docker ps

Squal78, 2024-09-09:
hello,

here is 
`
CONTAINER ID   IMAGE                                                        COMMAND                  CREATED         STATUS                  PORTS                                                                                                                          NAMES
c91fdfed400f   nspanelmanager/nspanelmanager-armv7:1.0.1                    "/bin/bash ./run_uws…"   12 hours ago    Up 12 hours             0.0.0.0:8000->8000/tcp, :::8000->8000/tcp                                                                                      addon_a5d2b728_nspanelmanager
`

Squal78, 2024-09-09:
Sure
`~ $ dpkg --print-architecture
armhf
`

tpanajott, 2024-09-09:
Could you send over the total output of `dpkg-architecture`? It should look something like the following:
```
DEB_BUILD_ARCH=arm64
DEB_BUILD_ARCH_ABI=base
DEB_BUILD_ARCH_BITS=64
DEB_BUILD_ARCH_CPU=arm64
DEB_BUILD_ARCH_ENDIAN=little
DEB_BUILD_ARCH_LIBC=gnu
DEB_BUILD_ARCH_OS=linux
DEB_BUILD_GNU_CPU=aarch64
DEB_BUILD_GNU_SYSTEM=linux-gnu
DEB_BUILD_GNU_TYPE=aarch64-linux-gnu
DEB_BUILD_MULTIARCH=aarch64-linux-gnu
DEB_HOST_ARCH=arm64
DEB_HOST_ARCH_ABI=base
DEB_HOST_ARCH_BITS=64
DEB_HOST_ARCH_CPU=arm64
DEB_HOST_ARCH_ENDIAN=little
DEB_HOST_ARCH_LIBC=gnu
DEB_HOST_ARCH_OS=linux
DEB_HOST_GNU_CPU=aarch64
DEB_HOST_GNU_SYSTEM=linux-gnu
DEB_HOST_GNU_TYPE=aarch64-linux-gnu
DEB_HOST_MULTIARCH=aarch64-linux-gnu
DEB_TARGET_ARCH=arm64
DEB_TARGET_ARCH_ABI=base
DEB_TARGET_ARCH_BITS=64
DEB_TARGET_ARCH_CPU=arm64
DEB_TARGET_ARCH_ENDIAN=little
DEB_TARGET_ARCH_LIBC=gnu
DEB_TARGET_ARCH_OS=linux
DEB_TARGET_GNU_CPU=aarch64
DEB_TARGET_GNU_SYSTEM=linux-gnu
DEB_TARGET_GNU_TYPE=aarch64-linux-gnu
DEB_TARGET_MULTIARCH=aarch64-linux-gnu
```

Squal78, 2024-09-12:
Sorry for my late answer, I was not at home
here is the command return

`pi@raspberrypi:~ $ dpkg-architecture
DEB_BUILD_ARCH=armhf
DEB_BUILD_ARCH_ABI=eabihf
DEB_BUILD_ARCH_BITS=32
DEB_BUILD_ARCH_CPU=arm
DEB_BUILD_ARCH_ENDIAN=little
DEB_BUILD_ARCH_LIBC=gnu
DEB_BUILD_ARCH_OS=linux
DEB_BUILD_GNU_CPU=arm
DEB_BUILD_GNU_SYSTEM=linux-gnueabihf
DEB_BUILD_GNU_TYPE=arm-linux-gnueabihf
DEB_BUILD_MULTIARCH=arm-linux-gnueabihf
DEB_HOST_ARCH=armhf
DEB_HOST_ARCH_ABI=eabihf
DEB_HOST_ARCH_BITS=32
DEB_HOST_ARCH_CPU=arm
DEB_HOST_ARCH_ENDIAN=little
DEB_HOST_ARCH_LIBC=gnu
DEB_HOST_ARCH_OS=linux
DEB_HOST_GNU_CPU=arm
DEB_HOST_GNU_SYSTEM=linux-gnueabihf
DEB_HOST_GNU_TYPE=arm-linux-gnueabihf
DEB_HOST_MULTIARCH=arm-linux-gnueabihf
DEB_TARGET_ARCH=armhf
DEB_TARGET_ARCH_ABI=eabihf
DEB_TARGET_ARCH_BITS=32
DEB_TARGET_ARCH_CPU=arm
DEB_TARGET_ARCH_ENDIAN=little
DEB_TARGET_ARCH_LIBC=gnu
DEB_TARGET_ARCH_OS=linux
DEB_TARGET_GNU_CPU=arm
DEB_TARGET_GNU_SYSTEM=linux-gnueabihf
DEB_TARGET_GNU_TYPE=arm-linux-gnueabihf
DEB_TARGET_MULTIARCH=arm-linux-gnueabihf
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

`
Thanks for the feedback, i will try to upgrade

tpanajott, 2024-09-13:
Thanks. Unfortunately it looks like you've ran into the same issue which we couldn't really work around last time. This is not an issue in newer raspbian versions.

We might be able to solve this by statically building the binary but I'm not sure. Anyway I've begun work to have that be the default when issuing a release.

gabylan12, 2024-09-23:
Curious, I am facing the same prolem

```
pi@living:~ $ uname -a
Linux living 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
```

nabossha, 2024-11-20:
same problem here for me, would love to help in resolving this issue :-)

`uname -a
Linux starknet 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`

Mistretzul, 2025-02-10:
> Thanks. Unfortunately it looks like you've ran into the same issue which we couldn't really work around last time. This is not an issue in newer raspbian versions.
>
> We might be able to solve this by [...]

I am running the latest Raspbian  (11) version and this still happens....

tpanajott, 2025-05-24:
This should be fixed as of latest beta as that is statically linked and not dynamically.

jamesmulcahy, 2026-10-25:
Backlog sweep: I did a batch pass over the open issues with Claude, checking each one against previous fixes and the current `devel` code.

Fixed in the v2 release builds, which link MQTTManager statically, so the relocation error no longer happens (as Tim mentioned in May 2025).

cc @tpanajott: I think this one can be closed.

tpanajott, 2026-10-25:
@jamesmulcahy yeah, this is the whole reason we do static linking. O just haven't closed it yet but will do. Please reopen if it's still an issue.

closed 2026-10-25 by tpanajott