NSPManager/NSPanelManager #181: Add support of blinds

tpanajott, 2024-09-10:
**Is your feature request related to a problem? Please describe.**
Users want to be able to control blinds/covers.

**Describe the solution you'd like**
A page where users can control blinds from the panel.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

tpanajott, 2024-09-10:
My beautiful thoughts on this topic:
[image]

cablesandcoffee, 2024-11-22:
[image]
Blinds page suggestion.

GalballyJ, 2025-01-08:
So plus one from me on here.

One thing to think about is how to best display a venetian blind style where you're tilting the blind rather than UP/Down. Perhaps you could have a blind type when adding the blind and you would get an appropriate UI element based on type?

For a standard roller blind i think you don't need the percentage type controls, a simple up/down/stop like restroom does (with a slider for fine control) should be more than enough

Ps love the project thank you for the work guys!

dsowinski, 2025-10-04:
+1 please make this happen

zonckerr75, 2026-01-03:
Hi guys,  thanks for all the improvements and hard work so far.  Any update on this particular aspect please? I've had to do some custom scenes for the blind controls but that's not ideal.   Sent you some coffees in advance to say thanks!

tpanajott, 2026-02-21:
> [@tpanajott](https://github.com/tpanajott) Just wanted to see where you guys were at with supporting blinds/covers? Thanks!

As always life has gotten in the way. Blinds/covers will probably be the next feature after thermostat functionality which we will release very shortly.