NSPManager/NSPanelManager #182: Rework "Room page" to show 4/8/12 entities

tpanajott, 2024-09-10:
Rework "Room page" to show a page of 4 entities if there are 4 or less entities registered in the room. If there are > 4 and <= 8 show a page with 8 entities on the page. If there are > 8 and <= 12 then show a page with 12 entities. If there are > 12 entities in the room, show a page with 8 entities and add buttons to toggle between different pages of entities for that room.

cablesandcoffee, 2024-11-22:
New pages, 4, 8, 12 devices
Device id on all pages are named the same as the old 12 devices page but just fewer.
Should probably change page names to Room1, Room2, Room3

[image]
[image]
[image]

tpanajott, 2025-03-23:
I belive this to be completed, will close.

closed 2025-03-23 by tpanajott