NSPManager/NSPanelManager #186: More scenes per room

oleyek, 2024-09-15:
I think in new version should be more space for scenes per room. Four scenes is not enough. Why can't there be eight scenes as in the case of light control?

cablesandcoffee, 2024-09-16:
Totally agree. We've talked about that.  I'll add this to the milestone.

cablesandcoffee, 2024-11-22:
My suggestion is that we reuse the updated room functionality with possibility to have 4, 8, 12 scenes depending on how many the user defines. We'll just add the save icon as a device on the room pages and set VISIBILITY when loading the page depending if user is accessing room page or scenespage.
[image]
[image]
[image]

tpanajott, 2025-03-23:
I suggest we start tracking this in issue #239 as that seems more appropriate but to give an answer to your suggestion @oleyek, this is basically implemented and somewhat ready for the next release.

I will close this as we don't need duplicate issues open for the same thing.

closed 2025-03-23 by tpanajott