NSPManager/NSPanelManager #188: Add toggle command to mqtt topics

psych0d0g, 2024-09-22:
as a user i have a battery powered 433mhz button near my bed to toggle the bedroom lights..
i can trigger custom mqtt messages by it using an sonoff rf_bridge.
to toggle the bedroom lights from this button it would be needed for the nspanel code to be able to react to toggle commands in addition to 0 and 1 sent to the corresponding mqtt queues.

tpanajott, 2024-09-22:
This should be easy enough to implement. I'm thinking that sending "2" would be a toggle instead.

psych0d0g, 2024-09-22:
that "should" work aswell for me :)

psych0d0g, 2024-10-16:
tested locally and working fine so far

tpanajott, 2025-04-07:
Issue has now been implemented and fixed in new and upcoming beta release. Will leave open until beta is published.

closed 2025-05-17 by tpanajott