NSPManager/NSPanelManager #189: Inside temperature sensor

psych0d0g, 2024-09-22:
**Is your feature request related to a problem? Please describe.**
Inside temperature sensor within the nspanel is really inaccurate, we can "calibrate" that with an offset, but during the heatup of the nspanel that is also just a really rough workaround

**Describe the solution you'd like**
I would like to be able to define an inside temperature sensor in addition to the outside temperature sensor to get the value from a more reliable source

tpanajott, 2024-11-17:
Hi, sorry I totally missed this. We are currently building on this in issue #205.

Anoxym, 2025-01-02:
We could use the calibration function of the NTC platform too 

# Example configuration entry
`sensor:
  - platform: ntc
    # ...
    calibration:
      - 10.0kOhm -> 25°C
      - 27.219kOhm -> 0°C
      - 14.674kOhm -> 15°C`

Here is the datasheet with values for the NTC MF52A (as described in the NSPanel teardown)

[image]

psych0d0g, 2025-01-08:
even with a calibrated internal sensor, the temp's would be off from reality, depending on how much the NsPanel itself warms up, and that is exponential in relation to the room temperature, so you can not just substract for example 5°C from the readings. using an external sensor's value for the display would be much more reliable

closed 2025-07-27 by tpanajott