NSPManager/NSPanelManager #191: Add a calendar view to the screensaver

psych0d0g, 2024-09-22:
I would like to suggest adding a calendar / tasks list to the screensaver, 
my idea would be to reduce the weather forecast to show only the next 2 days underneeth the current weather area, and then to the right under the time section, show upcoming tasks that should be possbile to be imported from a homeassistant calendar object.
This as an additional screensaver type would be nice to have

tpanajott, 2024-10-16:
What does @cablesandcoffee think för this idea, possible?

psych0d0g, 2024-10-16:
example usecase (for me at least): import the trash collection shedule ical from the trash company into homeassitant, and have the upcoming collection dates displayed on nearby nspanels in my rooms so my dumb brain dosnt forget to put the trashcan near the street ;)

tpanajott, 2024-10-16:
Yhea, sound like a reasonable request though I'm worried about the limited space.  I'm thinking we can only display one or two upcoming events and only the title for them.

psych0d0g, 2024-10-16:
maybe have it configurable at one point to either show a single upcoming event with a exerpt from the description, or multiple with only the title displayed

psych0d0g, 2024-10-16:
[image]
here is a REALLY bad mockup what i thought about :D

tpanajott, 2024-10-16:
It's better than a few of my mockups ^^ I'm sure @cablesandcoffee has a good idea on how to solve this. He always dose!

cablesandcoffee, 2024-10-18:
I think it's a good idea to add tasks in some way, something a lot of people use. Personally I would say we have a couple of other pages to finish before doing this but it's definitely possible. I'm thinking it could be hard to use the current weather screensaver and smash tasks in as well. If we want to do task stuff I think we should create a proper task page. Maybe we could have a swipe right on the screensaver to show the task page or something. Or a separate screensaver choice to only show tasks. Just some more thoughts, we're adding this to the to do list for sure.

psych0d0g, 2024-10-22:
maybe let the screensaver cycle to diffrent pages in a configurable way even?