NSPManager/NSPanelManager #194: Exception when adding Light to Room

johnderose, 2024-09-24:
**To Reproduce**
1. Configure NSPM, add one Panel, adjust some settings
2. Navigate to Rooms > Dummy Room (mine is named "1f_of" i.e. "1st Floor Office")
3. Select New Light
4. Exception: Max retries exceeded with url: /api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x778449661490>: Failed to establish a new connection: [Errno -2] Name or service not known')

**Screenshots**
- [nspm_exception.txt](https://github.com/user-attachments/files/17120981/nspm_exception.txt)

**Environment info (please complete the following information):**
 - Home Assistant 2024.9.3 (Docker)
 - Mosquitto (Docker)
 - NSPanelManager 1.0.2
 - NSPanel Firmware 0.1.152 (US)

**Additional context**
- Restarted Docker Containers; no change
- Restarted Host; no change
- HASS appears to be working normally

Please let me know if you I can help with troubleshooting. Thanks, JD

tpanajott, 2024-09-24:
Thank you for using NSPanel Manager. Are you running NSPanel Manager as an Home Assistant addon? I'm guessing so as you write that "HASS appears to be working normally". This looks like an exception while trying to get available entities via the "/api/states" on your Home Assistant host. Could you please confirm that settings for Home Assistant are applied in the "Settings" menu in the manager? There should be an address and token.

johnderose, 2024-09-24:
Hi Tim. Thanks for the quick reply. Yes, hass url and token are present in nspm, and hass api is responsive. The contents of /api/config are below... Kindly, JD

[hass_config.json](https://github.com/user-attachments/files/17121410/hass_config.json)

tpanajott, 2024-09-28:
Hi, sorry for the delay. What is the "Home Assistant Address" configured to in the "Settings" page in the manager? Given that you are running it as an addon it needs some special address (192.168.33.1 or something like that, I can't remember exactly) though that should be set as per default. Just double checking.

The "Name or service not known" error you got is basically "You've told me to connect to http://my-home-asssistant.com but I don't know that name or how to get there!".

jderose-net, 2024-10-03:
No problem... As I am running nspm under docker on the same bridged network as hass, I realized after what you said that maybe the address needed was the hass local address (vs. external interface). I applied the bridge ip, restarted the panel, and am able to access the list of rooms now. Thank you.

tpanajott, 2024-10-03:
> No problem... As I am running nspm under docker on the same bridged network as hass, I realized after what you said that maybe the address needed was the hass local address (vs. external interface). [...]

Nice. So the issue is resolved then?

jamesmulcahy, 2026-10-25:
Backlog sweep: I did a batch pass over the open issues with Claude, checking each one against previous fixes and the current `devel` code.

Resolved in the thread: the Home Assistant address was wrong. A friendlier error when HA can't be reached would still be nice; that can be a separate enhancement.

cc @tpanajott: I think this one can be closed.

closed 2026-11-07 by tpanajott