NSPManager/NSPanelManager #196: Reflash NSPanel through wifi

Rayeliint, 2024-10-17:
Not realy a feature request and not a bug.
This maybe a dumb question but I have 2 NS Panels that I purchased from another person a while back, they run NSPanel lovelace but I am having trouble with that and would like to try NS Panel manager instead.
Am I able to flash the panels to NSManager through Wifi using Tasmota or ESPHome or even logging into the panel directly through WiFi?
Pretty new to flashing as you can tell.

tpanajott, 2024-10-17:
Hi and welcome!

Unfortunately you need to flash via a serial flasher to get any other firmware as far as I'm aware. Just make sure you get a flasher that handles 3.3V, they are just a couple of dollars and should work fine. Think I bought mine about 5 years ago for something like 7$USD on AliExpress.

jamesmulcahy, 2026-10-25:
Backlog sweep: I did a batch pass over the open issues with Claude, checking each one against previous fixes and the current `devel` code.

The question was answered in the thread.

cc @tpanajott: I think this one can be closed.

closed 2026-10-25 by tpanajott