NSPManager/NSPanelManager #198: No Screensaver showing

Rayeliint, 2024-10-20:
**Describe the bug**
Hi there.
I have installed the firmware on a couple of NS Panels and all working except the screen saver part with Weather etc.
I am just getting a blank screen.
Global Settings I have screensaver timeout set to 30000, Screensaver Mode set to Show background (clock, date, Weather)
In entities I have latitude set to my home town and 24hr clock
On individual panels settings I have Screensaver  Use Global settings
Running Home Assistant 2024.10.3 on Raspberry Pi 4
NS Panel Manager BETA 0.1.46
NSPANEL version 0.1.152

aaronouthier, 2024-10-21:
You know, I didn't even think about that! I am having the same issue!
Home Assistant is also running on a RPI4. NSPM is running in an LCX container in Proxmox. 2 NS Panels, White, US version.

Backlight shuts off after screensaver delay is reached. I have brightness set to 80, and screensaver brightness set to 50. Those settings are set in HASSOS.

cablesandcoffee, 2024-10-21:
@Rayeliint just to confirm. what screensaver brightness have you set in global settings?

Rayeliint, 2024-10-21:
> @Rayeliint just to confirm. what screensaver brightness have you set in global settings?

OK, so I didnt pick that up!

Have changed to 100% and is working perfectly.

You can close this now.

closed 2024-10-21 by tpanajott