NSPManager/NSPanelManager #200: Minor issue with 12-hour time mode on screensaver.

aaronouthier, 2024-10-22:
**Describe the bug**
When NSPM is set to use 12-hour time, sunset is showing as 18:20 hours.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to NSPM
2. Click on "Entities->Weather and Time".
3. Scroll down to Time Format.
4. Select "12-Hour Clock"
5. Click "Save" at the bottom-right of the page.
6. On the main page, click the "Actions" drop-down, and then "Reboot all panels".
7. Observe sundown/sunset time is still expressed in 24-hour format.

**Expected behavior**
Sunrise/Sunup and Sunset/Sundown time format should reflect user setting.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment info (please complete the following information):**
 - Home Assistant - HASS OS on a RPi4
 - Chrome OS Flex v129.
 - NSPanelManager version [0.1.46 beta]
 - NSPanel Firmware version [0.1.152]

**Additional context**
NSPM installed as an LXC client on Proxmox 8.

aaronouthier, 2024-11-22:
Issue still ongoing.
Issue hasn't been acknowledged yet. Overlooked?

tpanajott, 2024-11-22:
Oh, I though I had answered this but apparently not. Ill take a look at it, should be easy enough to fix.

aaronouthier, 2024-11-22:
Thank you. I know you have much going on with this project.

cablesandcoffee, 2025-03-09:
Checked to see if this is fixed in the new devel firmware but it is still there. Maybe you already knew that but just as information.

tpanajott, 2025-04-23:
Just had a read through of existing issues, what does @aaronouthier  and @cablesandcoffee  think about the time for sunset/sunrise. Should the times be stated with AM/PM at the end or can the simply be omitted as self explanatory?

aaronouthier, 2025-04-23:
Thanks! I'll check the new build when I get a chance. Hopefully soon.

In most places, Sunrise is in the AM and Sunset is in the PM. However, cities near the Arctic & Antarctic Circles have weeks with either for example 4 hours of daylight or 4 hours of darkness near the Summer & Winter solstices. While they are the exception, not the rule, it would be nice to have AM/PM or at least A/P designations.

aaronouthier, 2025-04-23:
Yeah, I’m currently running devel build.

tpanajott, 2025-04-23:
Oh, wow, that's brave of you ^^

tpanajott, 2025-05-17:
Fixed in latest beta.

closed 2025-05-17 by tpanajott