NSPManager/NSPanelManager #204: Add climate panel

Oxygen54, 2024-11-13:
**Is your feature request related to a problem? Please describe.**
Since the NS Panel has a temperature sensor, it would be interesting to be able to use it as a thermostat.

**Describe the solution you'd like**
The best option is to use the Home Assistant Climate panel and have a page like this :

[image]

**Describe alternatives you've considered**
/

**Additional context**
All these settings would be available on Home Assistant to control power on/off, mode selection, ... as shown in the photo.

tpanajott, 2024-11-14:
I'm not quite sure what you are asking here, do you wish to have the NSPanel be able to control thermostats in Home Assistant or do you wish to be able to use the NSPanel as a thermostat? If so, there is already an issue tracking that #112

Oxygen54, 2024-11-14:
Yeah, be able to use the NS Panel as a thermostat, my plan is to deploy this panel in five room, currently I have thermostat in each room and my goal is to replace them with the NSPanel & NSPanel Manager. So yes as the same spirit like the #112 but update entities and not relays and uses an interface similar at the climate panel of Home Assistant

In example : 

- Panel_room_1_condition : "Heating to target", "Cooling to target", "Off"
- Panel_room_1_target_tmp : 25°C
- Panel_room_1_current_tmp : 21°C
- Panel_room_1_mode : Heat / Cool
- Panel_room_1_fan_speed : Low/Medium/high

Does that make sense ?

Oxygen54, 2024-11-14:
In my plan, use automation or script to control my air conditioner on HA, I will soon control my AC with an ESP32 with this project : https://github.com/JanM321/esphome-lg-controller

And instead of having the Home Assistant climate module, have it on the NS Panel like a thermostat.

tpanajott, 2024-11-14:
I'm not quite following why this can't already be done as you have the NSPanel temperature in HA already. Is there something I'm missing or is it simply the fact that you want to be able to control the HA thermostat from the NSPanel?

Oxygen54, 2024-11-14:
Yes, I want the ability to control the HA thermostat from each NS Panel in different rooms

kalwinskidawid, 2024-12-23:
I don't want to create another issue, because this is a simple question and is related to the climate. I have not found such information anywhere. 
@tpanajott  If the NSPanel disconnects from the manger for some reason, the thermostat continues to work properly (even if it goes into AP mode) ? The idea is to stay as autonomous as possible and not to overheat the floor, for example.

tpanajott, 2024-12-23:
> I don't want to create another issue, because this is a simple question and is related to the climate. I have not found such information anywhere.
> @tpanajott  If the NSPanel disconnects from the [...]

Yes, that is the plan!

GalballyJ, 2025-01-08:
So I have a similar but possibly simpler desire...

I already have a full tado setup integrated into HA. What I'd like to do is have a control on the nspanel as above that can display the HA temperature panel much the same as a normal dashboard and also allow the temperature to be adjusted. In my use case the nspanel would just be integrated into the HA temperature control and not need to be used as a sensor or anything like that.

tpanajott, 2025-04-03:
> @tpanajott  How are things going?

Well, we are getting ready for a new beta release in the next few weeks. This is not implemented yet into that but won't be far behind. I don't know if @cablesandcoffee has started work on the GUI parts yet?

cablesandcoffee, 2025-04-05:
Nope haven't created the page for climate but I have an idea in my mind. I think we just have to discuss what entities to support and how to implement it.

kalwinskidawid, 2025-05-30:
How is it going? Any success in pushing this topic forward?

aaronouthier, 2025-08-20:
Something to interface with Home Assistant's climate controls. I don't actually have HA setup like this just yet. Parts for it are coming Tomorrow (Thursday).

cablesandcoffee, 2025-08-21:
@aaronouthier got it! Then it's clear. I think there is another issue where the panel is supposed to be used as a thermostat where I guess you wire the relays to your heating. But this one is clear. We need a standard page to control the typical stuff related to heat pumps for example. :)

aaronouthier, 2025-08-21:
Great!With regard to the other issue then, with using the relays to act as a thermostat. This is probably possible, but one would only be able to control either heat OR cooling, as the fan connection is a separate wire from the other two, at least in the US. I believe this is common internationally however. As such, the Heat or AC would need to engage the Fan also. It might be possible to run both AC and Heat with 3 AC Triacs and 2 diodes, since Thermostats trigger on 24 volts AC, but the NSPanels use/send 120/240 AC (US or Europe standard). My electrical theory understanding is fundamental as best - that might or might not actually work. I do know a bit about thermostats, however. My Dad worked HVAC his whole career, and I picked up a few things. All-in-all, I'm not sure it is feasible to test all of this based solely on a layman's understanding of electronics. I'm sure the NSPanel software could be implemented, but I'm not sure if the "average Joe" could pull off wiring his or her home correctly. I should probably have mentioned this in the other issue...

tpanajott, 2025-08-21:
Hey, thanks for the information. I didn't quite know how the controls worked but as I remember the other enhsmcment it was ment for controlling under floor heating whish is basically on/off for that zone. I guess it could also be used to control stuff like baffels. 

> Great!
> 
> With regard to the other issue then, with using the relays to act as a thermostat. This is probably possible, but one would only be able to control either heat OR cooling, as the fan connection is a separate wire from the other two, at least in the US. I believe this is common internationally however. As such, the Heat or AC would need to engage the Fan also. It might be possible to run both AC/Heat and a FAN by jumping the Fan and Temp control source of choice with a wire and using a Triac between the NSPanel and the thermostat since Thermostats trigger on 24 volts AC, but the NSPanels use/send 120/240 AC (US/European standard). My electrical theory understanding is fundamental at best - that might or might not actually work. I do know a bit about thermostats, however, as my Dad worked HVAC his whole career, and I picked up a few things. All-in-all, I'm not sure it is feasible to test all of this based solely on a layman's understanding of electronics. I'm sure the NSPanel software could be implemented, but I'm not sure if the "average Joe" could pull off wiring his or her home correctly. I should probably have mentioned this in the other issue...
> 
> Also of note, is that one would need to use a separate temperature sensor for triggering, since the heat from operating the relays always throws off its temperature sensor.