NSPManager/NSPanelManager #205: Ability to define an OpenHab thing as indoor temperature source

sebastienvermeille, 2024-11-17:
**Is your feature request related to a problem? Please describe.**
I hate when two screens in the same room displays different temperatures.

The problem with thermometers is that if you have 5 different ones, they all will show a different temperature generally speaking... especially in IoT world.

My pellet stove provide one and a screen on it. I would like to keep this value as reference and use it as indoor temperature in the NSPanel screensaver.

**Describe the solution you'd like**
As an OpenHab user, I would like to be able to select an OpenHab device providing a more accurate temperature sensor of my choice rather than relying on the one provided by Sonoff NSPanel. (Same way as what exists for the outdoor temperature, we can provide an Openhab item as source)

**Describe alternatives you've considered**
We could alternatively define a custom MQTT topic, however I think we should follow the logic already in place which is to select an openhab item, home assistant item.

Providing a custom MQTT topic would be another feature that we could consider adding globally (also for the outdoor sensor) in the future (other PR).

**Additional context**
Here is how I plan to integrate this from a UX perspective:

[image]

WDYT? (I am willing to work on this feature if you agree)

tpanajott, 2024-11-17:
I thinks it's a good idea but I think that the user should also be able to select the item per room. So, either have the sensor selection in the room or on the specific panels, I would prefer the room page, what do you think?

Other than that I think it's a good idea and go for it!

sebastienvermeille, 2024-11-17:
Indeed it make totally sense: as the "inside temperature" is related to the nspanel's room it can't be a single temperature sensor for the whole.

Thinking about it: That would mean - the screen saver needs to have in context the current room (dunno if it's the case at the moment) and then be able to take the correct temperature measure.

I will probably be more aware of the flow while opening the source code :) Will come up with a PR and we can always refine, thank you for redirecting me to a better path!

tpanajott, 2024-11-17:
There is a variable for the current room and that is always set to the panels default room (where it is assigned) as soon as the panel goes to screensaver.

tpanajott, 2025-07-26:
Is this still something you want to work on or should I start working on it? It should be easy enough to implement to the next beta release.

tpanajott, 2025-07-27:
I took the liberty of implementing this while I still have the time (last day of semester) and since it's been open for many months now. It still needs more testing from en OpenHAB perspective since I don't have any temperature sensor implemented in OpenHAB to test with. Perhaps @cablesandcoffee has a sensor to test with?

sebastienvermeille, 2025-07-27:
thank you sorry I was not following this issue anymore

closed 2025-08-04 by cablesandcoffee