NSPManager/NSPanelManager #207: Migrate doc from PDF to github pages

sebastienvermeille, 2024-11-22:
**Is your feature request related to a problem? Please describe.**
Having to download the PDF doc each time I am willing to learn something about the project is a pain.
Also the edition of `manual.tex` is not straightforward, it require to regenerate the PDF etc.

**Describe the solution you'd like**
I would like to host the documentation as a web documentation using [Docusaurus](https://docusaurus.io/) and host it GitHub pages (free).

**Describe alternatives you've considered**
I tried a couple of other frameworks in the past such as Jekyll, docs.page.

While it would also work, Docusaurus is pretty straight forward, actively maintained and provide nice feature such as Search integration, PDF export.

**Additional context**

This has been validated on Discord while having a discussion with the Founders :+1: 

I will open a PR for this issue.

tpanajott, 2024-11-22:
Hey, nice work with the docs. A really big uplift from what there is currently. I'm just thinking though, currently there is basically two versions of the PDF. One, that is available for the stable/main branch and one that is for the beta software. Is there a way we can somehow mark certain things in the docs so that users are aware that they are currently only available in the beta version?

sebastienvermeille, 2024-11-25:
We have multiple solutions:

1) verisonize the doc (people can choose v1 or beta version)
As the doc says ... think twice if you really want to go this way:
https://docusaurus.io/docs/versioning

2) dedicated page/section of the doc (BETA)
2) add some tags to features that are in beta (same as tags on github PR)

tpanajott, 2024-11-25:
As discussed on Discord, we will work out how to mark beta-features before pushing new features into beta branch so as of now I'm going to merge this.

tpanajott, 2024-11-25:
PR #208 has been merged, what is left to do is setup DNS pointers from nspanelmanager.com to the GitHub pages. That is up to @cablesandcoffee to do as he is the manager of the web page.

closed 2024-12-04 by sebastienvermeille