NSPManager/NSPanelManager #216: After migrating to a new hardware with a new IP, NS panel does not connect

juancameijeiras, 2024-12-01:
**Describe the bug**
I have migrated my HA to another hardware with a different IP, I cannot connect anymore to my NSPANEL, I have checked the settings and rebooted the device physically without avail. Have tried with a standalone docker container with the same results, In the documentation there is no mention of how to proceed in this case

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment info (please complete the following information):**
 - Home Assistant  user
 - Browser : safari
 - NSPanelManager version 1.0.2
 - NSPanel Firmware version cannot check

aaronouthier, 2024-12-01:
Wow! I was just coming here to report the very same issue. Imagine my surprise to find this.

In my case, I'm not using Home Assistant, except for MQTT. ie, not running manager as an add-on. Safari on my iPad can't access my NSPanel after my docker system went down due to a temporary network reconfiguration. I am running the MosQuiTTo add-on in HAOS, which is used by my NSPanel and Manager.

Using Firefox on the same iPad works fine. Safari used to work just fine before my network change.

Suspect an incompatibility with Safari on iOS/iPadOS.

I am running beta firmware on my NSPanel.

My iPad is running iPadOS 18.1 on an iPad Air 4.

By the way, the OP didn't specify Safari on what hardware. iPhones, iPads, and Macs all have Safari installed.

tpanajott, 2024-12-01:
Strange that safari wouldn't work but it sounds like you solved it? If the manager has changed IP it needs to be updated in the settings of the manager.

aaronouthier, 2024-12-01:
Not exactly solved it, so much as found a work-around.
For what it's worth, Safari starts to connect, but it hangs, taking a long time with a blank screen, before giving a timeout. At least for me.