NSPManager/NSPanelManager #221: Screensaver goes back to default after Panel reboot

astritzeqiri, 2024-12-10:
**Describe the bug**
Screensaver goes back to default after Panel reboot. When you change the screensaver through homeassistant it changes on the panel, then if you reboot the panel it goes back to the default screensaver while on homeassistant it still has the changed one.
In the NSPanelManager it shows the default one.
I think the problem might be that the screensaver mode in the NSPanelManager config for that specific panel gets change back to the default one.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to HomeAssistant
2. Change the Panel screensaver mode
3. Go to NSPanel Manager
4. Go to Panel view
5. Click on Reboot
6. After panel reboots its back on the default screensaver mode

**Expected behavior**
The screensaver mode should be the one that it was changed to before the reboot.

**Screenshots**
[image]
[image]

**Environment info (please complete the following information):**
 - Home Assistant
 - Chrome
 - NSPanelManager version 1.0.4
 - NSPanel Firmware version [Visit NSPanel webpage to check] (CANT FIND IT)

tpanajott, 2024-12-10:
Hi and thank you for reporting. This is however expected as the setting you set in home assistant is only momentary. As soon as it reloads the config from the manager it will go back to that value. If you want to have one typ of screensaver permanently why not change it in the manager?

I'm trying to see if there's a use for setting it permanently or not through home assistant.

astritzeqiri, 2024-12-10:
Oh got it, didn't know that it is expected.
The thing is that I don't really think its a nice experience for people to allow this case to happen, to have a different screensaver mode on HA for the Panel to the one in the NSPanelManager.
For some reason it wasn't the behaviour I was expecting, if we allow the user to set the screensaver mode on HA then they would expect that to persist on the config aswell.

tpanajott, 2024-12-10:
Perhaps, pinging @sebastienvermeille and @cablesandcoffee to see what they think about it?

cablesandcoffee, 2024-12-14:
I think the manager is where you are supposed to manage your panels, HA or Openhab is not intended to manage the panels but through the different panel topics the users are able to automate different behaviors, for example changing the screensaver during different events or different times during the day. Same for screen brightness and so on. I personally don't want the standard panel settings to change in the manager when changing it from HA or Openhab. What I think would be good though is that if the panel is rebooted the settings that are on the panel on boot should be reflected in HA and Openhab, I guess that is not happening right now?

astritzeqiri, 2024-12-14:
I agree. Yes the panel settings are not being persisted to HA on panel restart. So the panel says it has weather screensaver and HA says panel has only clock screensaver.

tpanajott, 2025-04-04:
As implemented in the latest commit the changes in Home Assistant/OpenHAB will reflect the actual state in the firmware and not be lagging behind. Changing settings over MQTT will not be persistent and will be overwritten when the panel gets a config update from the manager. This can be changed later if a request for that is made. This will remain open for discoverability for current release but will be closed when the new beta release is out.