NSPManager/NSPanelManager #222: Ability to add other GPIO temperature sensor (DS18B20)

Anoxym, 2024-12-13:
**Is your feature request related to a problem? Please describe.**

Given that the current temperature sensor is not very accurate, it would be interesting to have the option on the web interface to select a different GPIO (e.g., GPIO23) to read the temperature.

**Describe the solution you'd like**

It would be useful to have a field allowing the selection of the default temperature sensor, either a Home Assistant sensor or a sensor connected to GPIO23. The best sensor I've found, which is easy to install and uses only one GPIO, is the DS18B20.

tpanajott, 2024-12-13:
Hmm, I like the idea but I'm not sure it's possible? To my knowledge there isn't any available GPIOs to connect to?

Anoxym, 2024-12-13:
From what I've seen on this teardown : https://blakadder.com/nspanel-teardown/

"You can notice that the headers are labelled BZ, RY1, RY2 and RY3. RY1 and RY2 are the existing relays and logic dictates RY3 is for the third relay. "

RY3 is GPIO23, it is therefore free

tpanajott, 2024-12-13:
Ah, nice! I didn't catch that on my first read through when first staring this, then, "small" project.

I'll look into adding DS18B20 support. As you say, it's a quite small sensor and only require 1 data pin.

Anoxym, 2024-12-13:
I ordered a few DS18B20 sensors and 4.7K pull-up resistors, I'll keep you updated on the results, but it should all be good.

--> in example with Tasmota : https://github.com/arendst/Tasmota/discussions/15737

tpanajott, 2024-12-13:
I think a have a few laying around from some custom PCBs. Just a matter of getting the code running.

tpanajott, 2024-12-13:
I'd just like to add that this feature will probably be implemented but it may be a while. Current beta and stable are based on a firmware written for the Arduino framework and we've pretty much reached what we can do with it. I'm currently working on a rewrite that will be based on ESP-IDF and will (mostly) be using protbuf instead of JSON.

Anoxym, 2024-12-25:
I confirm that the DS18B20 works on GPIO23. I connected to the test pad for GPIO23, a resistor pad for 3.3V, and the GND from the original NTC sensor. I used Tasmota to get the value in Home Assistant. It would be great to add an offset option to correct the sensor value if needed.

[image]

[image]

tpanajott, 2024-12-25:
> I confirm that the DS18B20 works on GPIO23. I connected to the test pad for GPIO23, a resistor pad for 3.3V, and the GND from the original NTC sensor. I used Tasmota to get the value in Home [...]

Thanks for wiring guide. This is planned for implementation but I've not yet had the time to get around to it. Regarding temperature offset there is already an option for that.

Anoxym, 2024-12-25:
No worries, it's just to keep track of it :)