NSPManager/NSPanelManager #223: failed to fetch logs from a5d2b728_nspanelmanager, FetchEvent.respondWith received an error: Load failed

MadeInFck, 2024-12-14:
**Describe the bug**
After installation of the home assistant add-on, just click on "start the add-on". Web UI is not available and checking logs, this error appears:
```
failed to fetch logs from a5d2b728_nspanelmanager, FetchEvent.respondWith received an error: Load failed
```

 Moreover, also noticed in logs:
```
[critical] [main.cpp:157] [29] No MQTT server or port configured! Will exit with code 1.
subprocess 29 exited by signal 6
```

**To Reproduce**
Purely straight forward installation as depicted in installation video shown on project website.

**Expected behavior**
Expected to access the UI panel of NSPanelManager in my browser.

**Screenshots**
If applicable, add screenshots to help e
[image]
[image]
xplain your problem.

**Environment info (please complete the following information):**
 - Home Assistant
 - Safari 18.2
 - NSPanelManager version 1.04
 - NSPanel Firmware version : unable to tell as I've flashed my device but can't access it without NSPanelManager

**Additional context**
Logs mention no MQTT server though I have Mosquitto broker running and working well with other add-ons in Home Assistant.

tpanajott, 2024-12-14:
Did you set the MQTT server and port in the settings of the manager? Looking at the logs it doesn't seem like it. It should have shown an initial setup guide when you first entered the web interface where you would be asked to enter those details.

MadeInFck, 2024-12-14:
Thanks for your response. I don't even have access to the settings of the manager. When I click on the web ui opening button, it fails. So I never get a change to set MQTT as the initial setup guide never shows up, unlike what is depicted in the video.

tpanajott, 2024-12-14:
Hm, that's sounds strange. The logs you referenced are expected on first setup and won't stop the web interface from working. Are you using it as an home assistant addon?

MadeInFck, 2024-12-14:
Just understood, the url opened by the web ui button is my nabu casa url. It works now, I have the initial setup panel on the local url. Thanks for that great project.

closed 2024-12-14 by MadeInFck