NSPManager/NSPanelManager #224: Host on UnRaid app store

sebastienvermeille, 2025-01-04:
**Is your feature request related to a problem? Please describe.**
As UnRAID user for my home automation system, I would like to be able to install NSPanelManager directly from the UnRAID app store. (Easier to setup rather than setting up a docker template manually)

**Describe the solution you'd like**
UnRAID provides an app store to install dockers via the user interface in "one click".
I would like to have a one click solution to install NSPanelManager in UnRAID and have it documented in the official documentation.

**Describe alternatives you've considered**
If NSPanelManager is not willing to support it officially, I can create a template and host it from my own repository for the unraid community this is not an issue.

The main concern here is, is NSPanelManager officially willing to maintain/support this unraid integration or not. (It is basically an XML file describing how to hook the docker into unraid. Here is an example: https://github.com/sebastienvermeille/unraid-docker-templates/blob/main/rika2mqtt/rika2mqtt.xml

**Additional context**

I can host it myself on my own repository for unraid as an alternative. (I was first willing to do it but well if we can document and maintain it in the official project it would be much better from a user perspective I suppose).

msebald, 2025-01-11:
Well, the template is pretty much straight forward. I intalled NSPanelManager on Unraid yesterday and this is my template. It can be found/created under `/boot/config/plugins/dockerMan/templates-user/my-NSPanelManager.xml`.

```
<?xml version="1.0"?>
<Container version="2">
  <Name>NSPanelManager</Name>
  <Repository>nspanelmanager/nspanelmanager-amd64:latest</Repository>
  <Registry>https://registry.hub.docker.com/r/nspanelmanager/nspanelmanager-amd64</Registry>
  <Network>bridge</Network>
  <MyIP/>
  <Shell>sh</Shell>
  <Privileged>false</Privileged>
  <Support/>
  <Project/>
  <Overview>NSPanel Manager is a custom software solution for the Sonoff NSPanel (not the NSPanel pro). The software is designed to be easy to use on a day to day basis and to easily manage multiple NSPanels around your home. The interface on the NSPanel itself has been designed to be intuitive to use for people of all ages and backgrounds.</Overview>
  <Category/>
  <WebUI>http://[IP]:[PORT:8000]/</WebUI>
  <TemplateURL/>
  <Icon>https://avatars.githubusercontent.com/u/140613208</Icon>
  <ExtraParams/>
  <PostArgs/>
  <CPUset/>
  <DateInstalled>1736550832</DateInstalled>
  <DonateText/>
  <DonateLink/>
  <Requires/>
  <Config Name="Data Storage" Target="/data" Default="" Mode="rw" Description="" Type="Path" Display="always" Required="false" Mask="false">/mnt/user/appdata/NSPanelManager/data</Config>
  <Config Name="Web Interface" Target="8000" Default="" Mode="tcp" Description="Port for Web Interface" Type="Port" Display="always" Required="false" Mask="false">8000</Config>
</Container>
```

The timezone is brought in by Unraid itself btw.

Finding/having NSPanelManager directly in the Unraid App Store would be a good thing though!

closed 2025-01-15 by tpanajott

sebastienvermeille, 2025-01-17:
[image]

App has been approved 👍