NSPManager/NSPanelManager #228: /MQTTManager/build/nspm_mqttmanager: error while loading shared libraries: /MQTTManager/build/libMQTTManager_WebHelper.so: unexpected reloc type 0x03

Mistretzul, 2025-02-10:
Hi guys,

I am getting this in the log : 

/MQTTManager/build/nspm_mqttmanager: error while loading shared libraries: /MQTTManager/build/libMQTTManager_WebHelper.so: unexpected reloc type 0x03

I am using Docker on Raspberry Pi 4. Docker Command:
docker run --name nspanelmanager   -e TZ=Europe/London   -v "/nspmdata/":"/data/"   -p 9200:8000   -d nspanelmanager/nspanelmanager-armv7:latest

 
Behaviour: I can access the WebServer but I cannot connect to MQTT. When checking the log I see the error above. I am not sure this is related, but I have checked the configuration and everything is correct. 

The NS Panel itself does connect to MQTT and transmits the correct request:

{"command":"register_request","mac_origin":"E4:65:B8:9B:C9:0C","friendly_name":"Living_Room_Wall_Switch","version":"0.1.152","md5_firmware":"","md5_data_file":"","md5_tft_file":"","address":"192.168.4.229"}

tpanajott, 2025-02-10:
Hi, please see issue #180

Mistretzul, 2025-02-10:
I have the latest Raspbian and this issue is still present

tpanajott, 2025-05-24:
This should be fixed as of latest beta as that is statically linked and not dynamically.

tpanajott, 2025-07-26:
Closing this issue as I've not heard back and theoretically it should be fixed. Please reopen if that is not the case.