NSPManager/NSPanelManager #229: "502 Bad Gateway" in Unraid App

Ksyuint, 2025-02-12:
After installing fresh from "Community Applications" in Unraid, get "502 Bad Gateway" in browser. Using default config. No forwarding to “Initial Setup”. Do I have to call it up manually?

tpanajott, 2025-02-12:
Hi and thank you for your interest in the project. I myself don't use unraid but perhaps @sebastienvermeille could help you? Do you have any logs?

Ksyuint, 2025-02-12:
Nice if I can help. Something is wrong with nginx:

> ...
No migrations to apply.
Starting Nginx
Restarting nginx: nginx[2025-02-12 16:28:37.888] [error] [WebHelper.cpp:68] [9] Failed to perform HTTP request to 'http://127.0.0.1:8000/api/get_mqtt_manager_config'. Got response data: ''
[2025-02-12 16:28:37.888] [error] [WebHelper.cpp:69] [9] curl_easy_perform() failed, got code: 7. Text interpretation: Couldn't connect to server
[2025-02-12 16:28:37.888] [error] [WebHelper.cpp:70] [9] Curl error buffer: Failed to connect to 127.0.0.1 port 8000 after 0 ms: Couldn't connect to server
[2025-02-12 16:28:37.888] [error] [mqtt_manager_config.cpp:116] [9] Failed to get config. Will try again.
.
Starting NSPanelManager web interface server
[2025-02-12 16:28:38.913] [debug] [mqtt_manager_config.cpp:111] [9] Got config data. Processing config.
[2025-02-12 16:28:38.914] [info] [mqtt_manager_config.cpp:127] [9] Got config from web manager, will process and load values.
[2025-02-12 16:28:38.914] [critical] [mqtt_manager_config.cpp:148] [9] Manager port not configured! Setting 0.
...

Port 8000 is used exclusively by NSPanelManager, there are no conflicts. It is a fresh installation without changing anything...

sebastienvermeille, 2025-02-13:
Hi @Ksyuint I will try to reproduce the issue

sebastienvermeille, 2025-02-13:
@Ksyuint I installed it on a fresh instance of Unraid and it works.

Please check your setup.

tpanajott, 2025-02-15:
I myself have during development received the "502 Bad Gateway" which usually refers to the Django-part of the application not starting but Nginx is running. We would need the full log output of the container/addon to look further into this.

thehijacker, 2025-02-28:
I too get 502 error. Fresh install with this docker compose:

```
  nspanelmanager:
    image: nspanelmanager/nspanelmanager-amd64
    container_name: nspanelmanager
    volumes:
      - /mnt/usb_1/Docker/NSPanelManager:/data/
    environment:
      - TZ=Europe/Ljubljana
    ports:
      - 8857:8000
```

When I visit the URL I get 502 error and in log:

`invalid request block size: 6190 (max 4096)...skip`

```
Setting /etc/timezone to Europe/Ljubljana
Will use timezone: Europe/Ljubljana
Running migrations...
Did not find a running MQTTManager, starting MQTTManager...
[2025-02-28 09:36:30.163] [info] [main.cpp:83] Starting MQTTManager.
[2025-02-28 09:36:30.163] [info] [main.cpp:91] /dev/shm has 67108864 bytes free.
[2025-02-28 09:36:30.163] [info] [main.cpp:96] Will log to /dev/shm/mqttmanager.log with a max size of 6710886 bytes.
[2025-02-28 09:36:30.166] [debug] [mqtt_manager.hpp:40] [9] Adding 'nspanel/mqttmanager/command' to the list of topics to subscribe to.
[2025-02-28 09:36:30.166] [debug] [mqtt_manager.hpp:40] [9] Adding 'nspanel/+/status' to the list of topics to subscribe to.
[2025-02-28 09:36:30.166] [debug] [mqtt_manager.hpp:40] [9] Adding 'nspanel/+/status_report' to the list of topics to subscribe to.
[2025-02-28 09:36:30.166] [info] [mqtt_manager_config.cpp:43] [9] Read timezone Europe/Ljubljana from /etc/timezone.
[2025-02-28 09:36:30.166] [info] [mqtt_manager_config.cpp:106] [9] Gathering config from web manager.
[2025-02-28 09:36:30.169] [error] [WebHelper.cpp:68] [9] Failed to perform HTTP request to 'http://127.0.0.1:8000/api/get_mqtt_manager_config'. Got response data: ''
[2025-02-28 09:36:30.169] [error] [WebHelper.cpp:69] [9] curl_easy_perform() failed, got code: 7. Text interpretation: Couldn't connect to server
[2025-02-28 09:36:30.169] [error] [WebHelper.cpp:70] [9] Curl error buffer: Failed to connect to 127.0.0.1 port 8000 after 0 ms: Couldn't connect to server
[2025-02-28 09:36:30.170] [error] [mqtt_manager_config.cpp:116] [9] Failed to get config. Will try again.
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, sessions, web
Running migrations:
  No migrations to apply.
Starting Nginx
Restarting nginx: nginx[2025-02-28 09:36:31.170] [error] [WebHelper.cpp:68] [9] Failed to perform HTTP request to 'http://127.0.0.1:8000/api/get_mqtt_manager_config'. Got response data: ''
[2025-02-28 09:36:31.170] [error] [WebHelper.cpp:69] [9] curl_easy_perform() failed, got code: 7. Text interpretation: Couldn't connect to server
[2025-02-28 09:36:31.170] [error] [WebHelper.cpp:70] [9] Curl error buffer: Failed to connect to 127.0.0.1 port 8000 after 0 ms: Couldn't connect to server
[2025-02-28 09:36:31.170] [error] [mqtt_manager_config.cpp:116] [9] Failed to get config. Will try again.
.
Starting NSPanelManager web interface server
*** Starting uWSGI 2.0.22 (64bit) on [Fri Feb 28 10:36:31 2025] ***
compiled with version: 12.2.0 on 07 December 2024 23:27:22
os: Linux-6.8.12-2-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-2 (2024-09-05T10:03Z)
nodename: 4a1931412d7b
machine: x86_64
clock source: unix
detected number of CPU cores: 16
current working directory: /usr/src/app
writing pidfile to /tmp/project-master.pid
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
chdir() to /usr/src/app/nspanelmanager/
your processes number limit is 125781
your memory page size is 4096 bytes
 *** WARNING: you have enabled harakiri without post buffering. Slow upload could be rejected on post-unbuffered webservers ***
detected max file descriptor number: 524288
lock engine: pthread robust mutexes
thunder lock: disabled (you can enable it with --thunder-lock)
uwsgi socket 0 bound to TCP address 127.0.0.1:8001 fd 3
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
Python version: 3.11.11 (main, Dec  4 2024, 20:38:25) [GCC 12.2.0]
[... 256 lines cut ...]
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:226] [41] Got mutex.
[2025-02-28 09:38:30.744] [debug] [room.cpp:16] [41] Room 1::Dummy room initialized.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:235] [41] Existing rooms updated.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:238] [41] Checking for removed rooms.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:268] [41] Updating NSPanel relay groups.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:274] [41] Existing relay groups updated.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:278] [41] Checking for removed relay groups.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:307] [41] Updating scenes.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:313] [41] Existing scenes updated.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:317] [41] Checking for removed scenes.
[2025-02-28 09:38:30.744] [info] [entity_manager.cpp:344] [41] Performing post init on 1 entities.
[2025-02-28 09:38:30.744] [debug] [entity_manager.cpp:346] [41] Performing PostInit on entity type 3 with id 1
[2025-02-28 09:38:30.744] [info] [entity_manager.cpp:350] [41] Total loaded NSPanels: 0
[2025-02-28 09:38:30.744] [info] [entity_manager.cpp:351] [41] Total loaded Entities: 1
[2025-02-28 09:38:30.745] [info] [main.cpp:153] [41] Starting Websocket Server on port 8002.
[2025-02-28 09:38:30.745] [critical] [main.cpp:157] [41] No MQTT server or port configured! Will exit with code 1.
[2025-02-28 09:38:30.745] [debug] [websocket_server.cpp:13] [49] Creating new ix::WebSocketServer.
terminate called without an active exception
subprocess 41 exited by signal 6
invalid request block size: 6228 (max 4096)...skip
invalid request block size: 6186 (max 4096)...skip
invalid request block size: 6236 (max 4096)...skip
invalid request block size: 6190 (max 4096)...skip

```

tpanajott, 2025-02-28:
> I too get 502 error. Fresh install with this docker compose:
>
> ```
>   nspanelmanager:
>     image: nspanelmanager/nspanelmanager-amd64
>     container_name: nspanelmanager
>     volumes:
>       - [...]

It seems that it tried to set a setting in the database before the DB was created. Have you tried restarting the container? This should only be an issue on first start though this should be fixed.

Ksyuint, 2025-02-28:
I have the same problem. The application was already installed once and worked. After deleting the application and reinstalling it, I had this problem (502). Are there perhaps temporary files somewhere (Docker volumes et cetra)? How can I restart the Setup or Config Wizard?

tpanajott, 2025-02-28:
> I have the same problem. The application was already installed once and worked. After deleting the application and reinstalling it, I had this problem (502). Are there perhaps temporary files somewhere (Docker volumes et cetra)? How can I restart the Setup or Config Wizard?

Well, in order to restart you need to delete the persistent database file (or simply move/rename it) that is stored on the persistent volume mapped to the `/data`-directory for the container. Can't recall the exact name but I belive that it should be something like `nspanelmanager_db.sqlite`.

thehijacker, 2025-02-28:
Content of directory:

```
root@docker-server:/mnt/usb_1/Docker/NSPanelManager# ll
total 236
drwxr-xr-x  2 root root   4096 Feb 28 10:57 ./
drwxrwxrwx 97 root root   4096 Feb 28 10:57 ../
-rw-r--r--  1 root root 229376 Feb 28 10:57 nspanelmanager_db.sqlite3
-rw-r--r--  1 root root     50 Feb 28 10:57 secret.key
```

And fresh log from container restart:

```
Setting /etc/timezone to Europe/Ljubljana
Will use timezone: Europe/Ljubljana
Running migrations...
Did not find a running MQTTManager, starting MQTTManager...
[2025-02-28 10:13:18.049] [info] [main.cpp:83] Starting MQTTManager.
[2025-02-28 10:13:18.050] [info] [main.cpp:91] /dev/shm has 67108864 bytes free.
[2025-02-28 10:13:18.050] [info] [main.cpp:96] Will log to /dev/shm/mqttmanager.log with a max size of 6710886 bytes.
[2025-02-28 10:13:18.052] [debug] [mqtt_manager.hpp:40] [9] Adding 'nspanel/mqttmanager/command' to the list of topics to subscribe to.
[2025-02-28 10:13:18.053] [debug] [mqtt_manager.hpp:40] [9] Adding 'nspanel/+/status' to the list of topics to subscribe to.
[2025-02-28 10:13:18.053] [debug] [mqtt_manager.hpp:40] [9] Adding 'nspanel/+/status_report' to the list of topics to subscribe to.
[2025-02-28 10:13:18.053] [info] [mqtt_manager_config.cpp:43] [9] Read timezone Europe/Ljubljana from /etc/timezone.
[2025-02-28 10:13:18.053] [info] [mqtt_manager_config.cpp:106] [9] Gathering config from web manager.
[2025-02-28 10:13:18.057] [error] [WebHelper.cpp:68] [9] Failed to perform HTTP request to 'http://127.0.0.1:8000/api/get_mqtt_manager_config'. Got response data: ''
[2025-02-28 10:13:18.057] [error] [WebHelper.cpp:69] [9] curl_easy_perform() failed, got code: 7. Text interpretation: Couldn't connect to server
[2025-02-28 10:13:18.057] [error] [WebHelper.cpp:70] [9] Curl error buffer: Failed to connect to 127.0.0.1 port 8000 after 0 ms: Couldn't connect to server
[2025-02-28 10:13:18.057] [error] [mqtt_manager_config.cpp:116] [9] Failed to get config. Will try again.
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, sessions, web
Running migrations:
  No migrations to apply.
Starting Nginx
Restarting nginx: nginx[2025-02-28 10:13:19.058] [error] [WebHelper.cpp:68] [9] Failed to perform HTTP request to 'http://127.0.0.1:8000/api/get_mqtt_manager_config'. Got response data: ''
[2025-02-28 10:13:19.058] [error] [WebHelper.cpp:69] [9] curl_easy_perform() failed, got code: 7. Text interpretation: Couldn't connect to server
[2025-02-28 10:13:19.058] [error] [WebHelper.cpp:70] [9] Curl error buffer: Failed to connect to 127.0.0.1 port 8000 after 0 ms: Couldn't connect to server
[2025-02-28 10:13:19.058] [error] [mqtt_manager_config.cpp:116] [9] Failed to get config. Will try again.
.
Starting NSPanelManager web interface server
*** Starting uWSGI 2.0.22 (64bit) on [Fri Feb 28 11:13:19 2025] ***
compiled with version: 12.2.0 on 07 December 2024 23:27:22
os: Linux-6.8.12-2-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-2 (2024-09-05T10:03Z)
nodename: 05e9713c42da
machine: x86_64
clock source: unix
detected number of CPU cores: 16
current working directory: /usr/src/app
writing pidfile to /tmp/project-master.pid
detected binary path: /usr/local/bin/uwsgi
!!! no internal routing support, rebuild with pcre support !!!
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
chdir() to /usr/src/app/nspanelmanager/
your processes number limit is 125781
your memory page size is 4096 bytes
 *** WARNING: you have enabled harakiri without post buffering. Slow upload could be rejected on post-unbuffered webservers ***
detected max file descriptor number: 524288
lock engine: pthread robust mutexes
thunder lock: disabled (you can enable it with --thunder-lock)
uwsgi socket 0 bound to TCP address 127.0.0.1:8001 fd 3
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
Python version: 3.11.11 (main, Dec  4 2024, 20:38:25) [GCC 12.2.0]
*** Python threads support is disabled. You can enable it with --enable-threads ***
Python main interpreter initialized at 0x73bea7c8f658
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
your server socket listen backlog is limited to 100 connections
[... 34 lines cut ...]
[2025-02-28 10:13:20.342] [debug] [entity_manager.cpp:222] [9] Trying to get room by id.
[2025-02-28 10:13:20.342] [debug] [entity_manager.cpp:224] [9] Got result. Trying to get mutex.
[2025-02-28 10:13:20.342] [debug] [entity_manager.cpp:226] [9] Got mutex.
[2025-02-28 10:13:20.342] [debug] [room.cpp:16] [9] Room 1::Dummy room initialized.
[2025-02-28 10:13:20.342] [debug] [entity_manager.cpp:235] [9] Existing rooms updated.
[2025-02-28 10:13:20.342] [debug] [entity_manager.cpp:238] [9] Checking for removed rooms.
[2025-02-28 10:13:20.343] [debug] [entity_manager.cpp:268] [9] Updating NSPanel relay groups.
[2025-02-28 10:13:20.343] [debug] [entity_manager.cpp:274] [9] Existing relay groups updated.
[2025-02-28 10:13:20.343] [debug] [entity_manager.cpp:278] [9] Checking for removed relay groups.
[2025-02-28 10:13:20.343] [debug] [entity_manager.cpp:307] [9] Updating scenes.
[2025-02-28 10:13:20.343] [debug] [entity_manager.cpp:313] [9] Existing scenes updated.
[2025-02-28 10:13:20.343] [debug] [entity_manager.cpp:317] [9] Checking for removed scenes.
[... cut ...]
[2025-02-28 10:13:20.343] [info] [entity_manager.cpp:351] [9] Total loaded Entities: 1
[2025-02-28 10:13:20.343] [info] [main.cpp:153] [9] Starting Websocket Server on port 8002.
[2025-02-28 10:13:20.343] [critical] [main.cpp:157] [9] No MQTT server or port configured! Will exit with code 1.
terminate called without an active exception
[2025-02-28 10:13:20.343] [debug] [websocket_server.cpp:13] [49] Creating new ix::WebSocketServer.
invalid request block size: 6259 (max 4096)...skip
invalid request block size: 6186 (max 4096)...skip
```

Same 502 error.

Ksyuint, 2025-02-28:
I had also completely deleted the folder beforehand and reloaded NSPanelManager. Unfortunately no improvement.
I deleted /mnt/user/appdata/nspanelmanager with all files and started docker again, still 502...

> Well, in order to restart you need to delete the persistent database file (or simply move/rename it) that is stored on the persistent volume mapped to the `/data`-directory for the container. Can't recall the exact name but I belive that it should be something like `nspanelmanager_db.sqlite`.

tpanajott, 2025-02-28:
> Content of directory:
>
> ```
> root@docker-server:/mnt/usb_1/Docker/NSPanelManager# ll
> total 236
> drwxr-xr-x  2 root root   4096 Feb 28 10:57 ./
> drwxrwxrwx 97 root root   4096 Feb 28 10:57 ../ [...]

Hmm, strange. It seems that the MQTTManager process has no problem talking to the Django part of the container according to the logs but it seems that going through Nginx doesn't work. Could you perhaps navigate into the container using something like Portainer or docker exec (`docker exec -it nspanemanager /bin/bash` or something like that) and try running `curl 127.0.0.1:8000`. That should return a lot of data from the Nginx->Django web interface communication. If that works then the problem is when trying to communicate into the container.

thehijacker, 2025-02-28:
Executing curl 127.0.0.1:8000 inside container I get a lot of HTML code. So that works. What else can I try to debug this?

tpanajott, 2025-02-28:
> Executing curl 127.0.0.1:8000 inside container I get a lot of HTML code. So that works. What else can I try to debug this?

Hmm, really strange. That would mean something goes wrong when the request comes from outside the container into the container. I really have no idea what it could be. @sebastienvermeille when you tried did you do so running the container with default port 8000 or did you change the port?

thehijacker, 2025-02-28:
Yes I also tried with default port 8000. I got exactly the same error message.

tpanajott, 2025-02-28:
> Yes I also tried with default port 8000. I got exactly the same error message.

Hmm, really strange. It seems that Nginx starts and runs normaly as it works from within the container but it seems that something goes wrong when it tries to handle connections from outside. Perhaps check if the are relevant logs in the `/var/log` directory for nginx and see if that reports any errors.

jamesmulcahy, 2026-10-25:
Backlog sweep: I did a batch pass over the open issues with Claude, checking each one against previous fixes and the current `devel` code.

The log in this thread has the cause: `invalid request block size: 6190 (max 4096)`. Cookies from other apps on the same host were overflowing uwsgi's 4 KB request-header buffer. Since v2.0 uwsgi runs with `--buffer-size 16384`, so this shouldn't happen on current versions. Clearing cookies for the host also works around it on older versions.

cc @tpanajott: I think this one can be closed.

tpanajott, 2026-10-25:
@jamesmulcahy i agree. In case this is still an issue, please reopen it.

closed 2026-10-25 by tpanajott