NSPManager/NSPanelManager #230: Failed to get file size for TFT flashing

AndrewsHappy, 2025-02-17:
I have flashed my NSPanel, and got as far as seeing it on the NSPanel Manager GU.  The screen of the NSPanel, is still showing the sonoff logo GUI.  When I go into the settings of the NSPanel, I can hit the update firmware, the log shows the following : 

04:45:07 PM | ERROR | NSPanel.cpp:681 f(._initTFTUpdate) Failed to get file size for TFT flashing. Will try again.
04:45:07 PM | ERROR | HttpLib.cpp:14 f(.GetFileSize) Failed to retrive file size for URL 'http://homeassistant.local:8000/download_tft_eu'. Got return code: -1
04:44:56 PM | INFO | NSPanel.cpp:676 f(._initTFTUpdate) Will download TFT file from: http://homeassistant.local:8000/download_tft_eu
04:44:56 PM | DEBUG | NSPanel.cpp:658 f(._initTFTUpdate) Got comok: comok 2,30614-0,NX4832F035_011C,102,61744,086634019E96C372,16777
04:44:56 PM | DEBUG | NSPanel.cpp:649 f(._initTFTUpdate) Sending connect to panel
04:44:52 PM | INFO | NSPanel.cpp:635 f(._initTFTUpdate) Force restarting screen via power switch.
04:44:52 PM | INFO | NSPanel.cpp:618 f(._initTFTUpdate) Trying to initialize TFT update at baud 115200
04:44:52 PM | INFO | NSPanel.cpp:747 f(._updateTFTOTA) _updateTFTOTA Started.
04:44:52 PM | INFO | NSPanel.cpp:563 f(._taskUpdateTFTConfigOTA) Starting TFT update...
04:44:52 PM | INFO | InterfaceManager.cpp:56 f(.stop) InterfaceManager stopped.

If I put http://homeassistant.local:8000/download_tft_eu into a web browser, it will try to download "download_tft_eu" but stops it by saying "insecure download blocked"  I can force it to download, then if I try to put this file into the www of homeassistant, it just vanishes!

Any ideas ?

I h

tpanajott, 2025-02-17:
Hi, for "manager address" in the settings of the manager you need to enter the IP for homeassistant.local and not the address. Change that, reboot the panel and try again. Should work then!

tpanajott, 2025-02-17:
Hey, happy to help! Glad you got it running :) Perhaps we need to change the working of that one so "manager IP" or something like that.

tpanajott, 2025-03-08:
Issue has been resolved, closing.

closed 2025-03-08 by tpanajott