NSPManager/NSPanelManager #231: Screen saver remains blank

AndrewsHappy, 2025-02-17:
I am still learning this fantastic bit of software, I am struggling to get the screen saver set up, at the moment it just goes blank after 30 seconds. I have changed the activation timeout to various values, but the screen blanks out at the same interval.  I have changed the screensaver mode, but no matter what I change it to, it still goes to just a blank screen.

After each change, I am hitting the save button, and then rebooting the nspanel. Even tried to hit the firmware update button. But what ever I do, nothing seems to get the screen saver to change from a blank screen.

What step am I missing ?

tpanajott, 2025-02-17:
Well, you are not alone, this is something a lot of users miss. I'm guessing you still have screensaver brightness set to 0?

AndrewsHappy, 2025-02-17:
Easy when you know how! It was set on Global, I was not sure what Global was or what the scale was, which is why didn't think to change it.  Anyway, I have the screen saver working now.  Once again, thank you.

closed 2025-02-17 by AndrewsHappy

tpanajott, 2025-02-17:
Glad you got it working. Global basically means to use whatever is set in the "Settings" menu in the top menu.