NSPManager/NSPanelManager #232: Left and Right button assignment

AndrewsHappy, 2025-02-19:
As I understand it, you can have as many different light entities, which are put into either the ceiling or table group. 
For example

- Twilight
- Floor lights
- Display
All go into the Table light category

- Rope light
- Top Light
All go into the Ceiling light category.

I can then either change the brightness of them all at the same time, or hold either the ceiling or Table while the other is changed.  I can even change each individual light. 

What I do not get, is the point of assigning one of the switches to for example Twilight. All I can see is, that it makes it easy to just turn on the Twilight light rather than work my way through the menu.  Is there a way to have this as another category ie, Ceiling Table Button 1 Button 2

AndrewsHappy, 2025-02-19:
OK! I have worked it out (well, work in progress) I need to use MQTT with an automation to sort this out.... I must say this NSPanel is really making me sort out my home assistant bugs etc...  Great Product

tpanajott, 2025-03-08:
Hey, sorry for being late but it seems you figured it out by yourself. Thanks you using NSPanel Manager, I'm going to close this issue as it has been resolved.

As for binding a button to a light, it's a way to control a single light easily. For example, you may want to frequently toggle only a single light in the ceiling group and by placing that on a physical button it can be easily accomplished.

closed 2025-03-08 by tpanajott