NSPManager/NSPanelManager #234: *DEVEL* No status is received on the panel on boot

cablesandcoffee, 2025-02-22:
**Describe the bug**
All thought lights are on everything seems off on the panel. No status is received.

tpanajott, 2025-02-22:
This should now be fixed. It was caused by the same issue that solved #235

cablesandcoffee, 2025-02-22:
Partly fixed I think. If the manager is just started no status is sent out to panels on boot. If I change some light settings and reboot the panel only those things that has happened since the manager was turned on is sent out to panel on boot. If I restart the manager after this and reboot a panel no status at all is sent out to the panel.

So it seems like status of lights is not collected to the manager when the manager is started so it can send that data out to the panels?

tpanajott, 2025-02-25:
Think I managed to fix it. Please retry.

cablesandcoffee, 2025-02-27:
Yes there seem to be a lot of errors actually:

`
[2025-02-26 21:04:23.476] [debug] [light.cpp:40] [3445] Light 1::Bed left base loaded, can dim: yes, can color temp: yes, can_rgb: yes. Controlled from main page? Yes. Entities page ID 1 in slot 0. [2025-02-26 21:04:23.483] [error] [openhab_light.cpp:344] [3445] Light 1::Bed left got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedLeft_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.485] [error] [openhab_light.cpp:368] [3445] Light 1::Bed left got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedLeft_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.485] [debug] [openhab_light.cpp:67] [3445] Loaded light 1::Bed left. [2025-02-26 21:04:23.485] [info] [entity_manager.cpp:159] [3445] Light 1::Bed left was found in database but not in config. Creating light. [2025-02-26 21:04:23.486] [debug] [light.cpp:50] [3445] Loading light 2::Bed Right. [2025-02-26 21:04:23.486] [debug] [light.cpp:40] [3445] Light 2::Bed Right base loaded, can dim: yes, can color temp: yes, can_rgb: yes. Controlled from main page? Yes. Entities page ID 1 in slot 1. [2025-02-26 21:04:23.490] [error] [openhab_light.cpp:344] [3445] Light 2::Bed Right got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedRight_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.492] [error] [openhab_light.cpp:368] [3445] Light 2::Bed Right got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedRight_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.492] [debug] [openhab_light.cpp:67] [3445] Loaded light 2::Bed Right. [2025-02-26 21:04:23.492] [info] [entity_manager.cpp:159] [3445] Light 2::Bed Right was found in database but not in config. Creating light. [2025-02-26 21:04:23.493] [debug] [light.cpp:50] [3445] Loading light 3::Desktop. [2025-02-26 21:04:23.493] [debug] [light.cpp:40] [3445] Light 3::Desktop base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 2. [2025-02-26 21:04:23.497] [error] [openhab_light.cpp:344] [3445] Light 3::Desktop got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item DesktopBlue_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.497] [debug] [openhab_light.cpp:67] [3445] Loaded light 3::Desktop. [2025-02-26 21:04:23.497] [info] [entity_manager.cpp:159] [3445] Light 3::Desktop was found in database but not in config. Creating light. [2025-02-26 21:04:23.498] [debug] [light.cpp:50] [3445] Loading light 4::Ceiling. [2025-02-26 21:04:23.498] [debug] [light.cpp:40] [3445] Light 4::Ceiling base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 3. [2025-02-26 21:04:23.502] [error] [openhab_light.cpp:344] [3445] Light 4::Ceiling got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item RetroGrandma_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.502] [debug] [openhab_light.cpp:67] [3445] Loaded light 4::Ceiling. [2025-02-26 21:04:23.502] [info] [entity_manager.cpp:159] [3445] Light 4::Ceiling was found in database but not in config. Creating light. [2025-02-26 21:04:23.503] [debug] [light.cpp:50] [3445] Loading light 16::Mirror. [2025-02-26 21:04:23.503] [debug] [light.cpp:40] [3445] Light 16::Mirror base loaded, can dim: no, can color temp: no, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 4. [2025-02-26 21:04:23.505] [debug] [openhab_light.cpp:67] [3445] Loaded light 16::Mirror. [2025-02-26 21:04:23.505] [info] [entity_manager.cpp:159] [3445] Light 16::Mirror was found in database but not in config. Creating light. [2025-02-26 21:04:23.506] [debug] [light.cpp:50] [3445] Loading light 17::Window. [2025-02-26 21:04:23.506] [debug] [light.cpp:40] [3445] Light 17::Window base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 5. [2025-02-26 21:04:23.510] [error] [openhab_light.cpp:344] [3445] Light 17::Window got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item TheBlue_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.510] [debug] [openhab_light.cpp:67] [3445] Loaded light 17::Window. [2025-02-26 21:04:23.510] [info] [entity_manager.cpp:159] [3445] Light 17::Window was found in database but not in config. Creating light. [2025-02-26 21:04:23.511] [debug] [light.cpp:50] [3445] Loading light 5::Sink. [2025-02-26 21:04:23.511] [debug] [light.cpp:40] [3445] Light 5::Sink base loaded, can dim: yes, can color temp: no, can_rgb: no. Controlled from main page? Yes. Entities page ID 2 in slot 0. [2025-02-26 21:04:23.513] [debug] [openhab_light.cpp:67] [3445] Loaded light 5::Sink. [2025-02-26 21:04:23.513] [info] [entity_manager.cpp:159] [3445] Light 5::Sink was found in database but not in config. Creating light. [2025-02-26 21:04:23.514] [debug] [light.cpp:50] [3445] Loading light 6::Stove. [2025-02-26 21:04:23.514] [debug] [light.cpp:40] [3445] Light 6::Stove base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Ye [... cut ...] can_rgb: yes. Controlled from main page? Yes. Entities page ID 4 in slot 0. [2025-02-26 21:04:23.523] [error] [openhab_light.cpp:344] [3445] Light 12::Ledstrip mirror got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item LedstripBathroom_ColorTemperature though current color temperature state was not a number. [2025-02-26 21:04:23.525] [error] [openhab_light.cpp:368] [3445] Light 12::Ledstrip mirror got OpenHAB data from OpenHAB via initial ItemStateFetched event fo
`

tpanajott, 2025-02-27:
Hm, that seems correct though it seems to only affect color temperature so brightness should still work.

cablesandcoffee, 2025-02-27:
Okay, maybe the logs from a manager restart is interesting. Is there anything when you start the manager that shows if the manager collects values succesfully from openhab?

Here is a log from restarting the manager:

`Killing running MQTTManager
Did not find a running MQTTManager, starting MQTTManager...
Starting a new mqtt_manager
[pid: 239|app: 0|req: 39/205] 172.17.0.1 () {54 vars in 977 bytes} [Thu Feb 27 20:49:24 2025] POST /api/restart_mqtt_manager => generated 16 bytes in 2 msecs (HTTP/1.1 200) 6 headers in 204 bytes (1 switches on core 0)
[2025-02-27 20:49:24.090] [info] [main.cpp:94] Starting MQTTManager.
[2025-02-27 20:49:24.090] [info] [main.cpp:102] /dev/shm has 61145088 bytes free.
[2025-02-27 20:49:24.090] [info] [main.cpp:107] Will log to /dev/shm/mqttmanager.log with a max size of 6114508 bytes.
[2025-02-27 20:49:24.090] [info] [main.cpp:157] [785] SQLite3 seems to be compiled with threadsafe mutexes. Setting: 1
[2025-02-27 20:49:24.090] [info] [mqtt_manager_config.cpp:59] [785] Read timezone Europe/Stockholm from /etc/timezone.
[2025-02-27 20:49:24.090] [info] [mqtt_manager_config.cpp:62] [785] Loading MQTT Manager settings.
[2025-02-27 20:49:24.096] [info] [weather.cpp:28] [785] No running instance of weather manager found. Will start new thread.
[2025-02-27 20:49:24.096] [info] [ipc_handler.cpp:13] [787] Starting ZMQ IPCHandler.
[2025-02-27 20:49:24.097] [info] [mqtt_manager.cpp:92] [785] Reconnecting MQTT as settings has changed.
[2025-02-27 20:49:24.097] [info] [main.cpp:174] [785] Starting Websocket Server on port 8002.
[2025-02-27 20:49:24.097] [info] [mqtt_manager.cpp:40] [791] Starting MQTT message processing thread.
[2025-02-27 20:49:24.097] [info] [main.cpp:177] [785] Config loaded. Starting components.
[2025-02-27 20:49:24.097] [info] [mqtt_manager.cpp:117] [791] Appending mac data from eth0.
[2025-02-27 20:49:24.097] [info] [main.cpp:183] [785] Home Assistant and/or OpenHAB initialized. Will load entities.
[2025-02-27 20:49:24.097] [debug] [home_assistant_manager.cpp:28] [795] Initializing Home Assistant Manager component.
[2025-02-27 20:49:24.097] [info] [mqtt_manager.cpp:141] [791] Will connect to MQTT with Client name NSPMMQTT_02_42_ac_11_ff_ff
[2025-02-27 20:49:24.097] [info] [mqtt_manager.cpp:152] [791] Will connect to MQTT via tcp://192.168.32.202:1883 as client NSPMMQTT_02_42_ac_11_ff_ff
[2025-02-27 20:49:24.097] [debug] [openhab_manager.cpp:20] [796] Initializing Openhab Manager component.
[2025-02-27 20:49:24.097] [debug] [websocket_server.cpp:13] [792] Creating new ix::WebSocketServer.
[2025-02-27 20:49:24.097] [debug] [websocket_server.cpp:19] [792] Start listening for websocket connection on 0.0.0.0:8002.
[2025-02-27 20:49:24.097] [debug] [websocket_server.cpp:23] [792] Successfully bound to 0.0.0.0:8002
[2025-02-27 20:49:24.098] [error] [home_assistant_manager.cpp:44] [795] No Home Assistant address configured. Will not continue to load Home Assistant component.
[2025-02-27 20:49:24.098] [info] [openhab_manager.cpp:86] [796] Will connect to OpenHAB with new settings. Server address: http://192.168.32.202:8080
[2025-02-27 20:49:24.098] [debug] [openhab_manager.cpp:93] [796] Keepalive thread exited, start new connection.
[2025-02-27 20:49:24.098] [info] [openhab_manager.cpp:47] [796] Will connect to Openhab websocket at ws://192.168.32.202:8080/ws
[2025-02-27 20:49:24.098] [debug] [openhab_manager.cpp:48] [796] Appending Openhab access token to url.
[2025-02-27 20:49:24.098] [debug] [openhab_manager.cpp:68] [796] OpenHAB connect finished.
[2025-02-27 20:49:24.098] [info] [openhab_manager.cpp:261] [802] Started Openhab keepalive thread.
[2025-02-27 20:49:24.102] [info] [openhab_manager.cpp:106] [801] Connected to Openhab websocket.
[2025-02-27 20:49:24.197] [debug] [weather.cpp:83] [786] Successfully received new weather forecast from Open Meteo. Will process new data.
[2025-02-27 20:49:24.198] [debug] [weather.cpp:156] [786] Loaded forecast for 5 days.
[2025-02-27 20:49:24.198] [debug] [weather.cpp:257] [786] Sending new weather data out on topic nspanel/mqttmanager_192.168.32.201/status/weather.
[2025-02-27 20:49:24.198] [info] [mqtt_manager.cpp:174] [791] Established connection to MQTT server.
[2025-02-27 20:49:24.198] [debug] [mqtt_manager.cpp:177] [791] Subscribing to registered MQTT topics.
[2025-02-27 20:49:24.199] [debug] [mqtt_manager.hpp:48] [785] Adding 'nspanel/mqttmanager/command' to the list of topics to subscribe to.
[2025-02-27 20:49:24.199] [debug] [mqtt_manager.hpp:51] [785] MQTT is connected, subscribing to MQTT topic 'nspanel/mqttmanager/command'.
[2025-02-27 20:49:24.240] [debug] [mqtt_manager.hpp:48] [785] Adding 'nspanel/+/status' to the list of topics to subscribe to.
[2025-02-27 20:49:24.240] [debug] [mqtt_manager.hpp:51] [785] MQTT is connected, subscribing to MQTT topic 'nspanel/+/status'.
[2025-02-27 20:49:24.241] [info] [entity_manager.cpp:62] [785] Loading config...
[2025-02-27 20:49:24.241] [debug] [weather.cpp:85] [786] Weather data processed.
[2025-02-27 20:49:24.244] [info] [entity_manager.cpp:174] [785] Loading 3 scenes.
[2025-02-27 20:49:24.245] [debug] [openhab_scene.cpp:30] [785] Loaded OpenHAB scene 1::Sovläge.
[2025-02-27 20:49:24.245] [info] [entity_manager.cpp:198] [785] Scene 1::Sovläge was found in database but not in config. Creating scene.
[2025-02-27 20:49:24.245] [debug] [openhab_scene.cpp:30] [785] Loaded OpenHAB scene 2::Vakenläge.
[... cut ...]
[2025-02-27 21:49:27.102] [error] [nspanel.cpp:520] [793] Stacktrace: Dynamic exception type: nlohmann::json_abi_v3_11_3::detail::parse_error
std::exception::what: [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - invalid literal; last read: '<U+0018>'

[2025-02-27 21:49:27.102] [info] [nspanel.cpp:524] [793] Failed to parse data. Will try as protobuf.
[pid: 241|app: 0|req: 53/206] 172.17.0.1 () {50 vars in 930 bytes} [Thu Feb 27 20:49:27 2025] GET /htmx/component/nspanel_warnings/index/11 => generated 748 bytes in 21 msecs (HTTP/1.1 200) 6 headers `

tpanajott, 2025-03-08:
I believe this could be closed?

cablesandcoffee, 2025-03-08:
yes closing! :)

closed 2025-03-08 by cablesandcoffee