NSPManager/NSPanelManager #237: *DEVEL* FIRST PAGE - Toggling between rooms sometimes stops working

cablesandcoffee, 2025-02-22:
**Describe the bug**
Haven't seen yet what causes it but quite frequently the toggle between rooms button stops working and you cannot toggle between rooms. I'll keep investigating...

cablesandcoffee, 2025-02-22:
Log from when it stopped working. Error message when trying to toggle.

`Time 	Level 	Message
09:06:09	ERROR	(483742) RoomManager: Tried going to another room but no rooms exist in current config.
09:06:08	ERROR	(482402) RoomManager: Tried going to another room but no rooms exist in current config.
09:06:07	ERROR	(481082) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:56 	ERROR 	(470152) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:56 	ERROR 	(469932) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:55 	ERROR 	(469752) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:55 	ERROR 	(469592) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:55 	ERROR 	(469282) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:55 	ERROR 	(468902) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:54 	ERROR 	(468092) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:53 	ERROR 	(467422) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:52 	ERROR 	(466282) RoomManager: Tried going to another room but no rooms exist in current config.
09:05:51 	ERROR 	(465402) RoomManager: Tried going to another room but no rooms exist in current config.
09:02:20 	ERROR 	(253852) event: handler already registered, overwriting
09:01:49 	DEBUG 	(223052) EntitiesPage: Unknown component ID of touch event in items8. ID: 9
09:01:49 	DEBUG 	(222942) HomePage: Sending command to toggle entity in slot 4 from entity page with ID 3.
09:01:48 	DEBUG 	(222682) EntitiesPage: Unknown component ID of touch event in items8. ID: 1
09:01:48 	DEBUG 	(222652) HomePage: Sending command to toggle entity in slot 73 from entity page with ID 3.
09:01:48 	ERROR 	(222482) Nextion: Unknown event data from Nextion: $`

cablesandcoffee, 2025-02-22:
It seems to stop when you restart the manager.

tpanajott, 2025-02-22:
Known issue that I haven't had the time to look into. Will keep open to track progress.

tpanajott, 2025-03-15:
I think this has been fixed. The issue was that the manager overloaded the panel with to many config updates for it to handle. I've not seen this since that issue was fixed.

cablesandcoffee, 2025-03-15:
sounds good, closing

closed 2025-03-15 by cablesandcoffee