NSPManager/NSPanelManager #238: *DEVEL* FIRST PAGE - Entering room sends user to wrong room

cablesandcoffee, 2025-02-22:
**Describe the bug**
Entering the default room works the first time you try it. If you switch room from the room page and go back to the first page no matter what room you toggle to you will always get back in to the room that was last showed on the room page when trying to enter.

tpanajott, 2025-03-15:
This doesn't seem to be the case any longer if I understand the issue correctly? I stand on my default room (Office) and navigate to the entities-page for that room. I then navigate (within the entities pages) to another room (test) and when I then exit the entities-page for the room test the Home Page is still on room test as expected.

tpanajott, 2025-03-15:
I managed to find what the issue was. Should now be fixed.

cablesandcoffee, 2025-03-15:
Still seeing this issue. With a just freshly started panel I get the wrong room.

My Default room is Livingroom
When the panel has started is says Livingroom on the room button
Pressing Livingroom and I get Entryway.

I have the following rooms:
Bedroom
Kitchen
Livingroom
Bathroom
Entryway
test

Panel Log when pressing Livingroom button on panel

'''23:13:27	DEBUG	(237494) EntitiesPage: Received new data. Will update display.
23:13:27	DEBUG	(237494) RoomManager: Received new entities page state update.
23:13:27	DEBUG	(237474) RoomManager: Received new home page state update.
23:13:27	DEBUG	(237404) RoomManager: Request to navigate to entities page ID 5
23:13:27	DEBUG	(237394) RoomManager: Subscribing to room topic nspanel/mqttmanager_192.168.32.201/room/5/state.
23:13:27	DEBUG	(237394) RoomManager: Request to navigate to room page ID 5
23:13:27	DEBUG	(237384) RoomManager: Request to navigate to entities page ID 4
23:13:27	DEBUG	(237384) RoomManager: Subscribing to room topic nspanel/mqttmanager_192.168.32.201/room/4/state.
23:13:27	DEBUG	(237374) RoomManager: Request to navigate to room page ID 4
23:13:27	DEBUG	(237364) RoomManager: Request to navigate to entities page ID 18
23:13:24	DEBUG	(234874) HomePage: Updating displayed values on HomePage.'''

tpanajott, 2025-03-23:
I though I had fixed this, which I had, but then apparently introduced another bug. I'm pretty sure this should be working as of latest commit.

cablesandcoffee, 2025-03-28:
Yepp this works now! Nice! :) closing

closed 2025-03-28 by cablesandcoffee