NSPManager/NSPanelManager #239: *DEVEL* Can't enter scenes page

cablesandcoffee, 2025-02-22:
**Describe the bug**
Maybe not implemented yet but cannot enter scenes page.

tpanajott, 2025-02-22:
Yup, not implemented as of yet. Will keep issue open as to track progress.

tpanajott, 2025-03-15:
How do we wish to handle scene pages? With entities page it's pretty straight forward as pretty much every room will have a page, why would you otherwise have a room?

Anyway, perhaps not every room will have scenes, so, what should we do with the user taps the "Scene" button for a room that has 0 rooms. Do nothing? Show an empty page? What page, 4, 8 or 12? Should we go to the next available scenes page, scanning forward within the available rooms? That might be confusing though.

cablesandcoffee, 2025-03-15:
I'm thinking just show an empty 4 scenes pages? For a new user it could be good to be sent to something if they just play around and want to see what that settings icon does. They will realize there is a scenes functionality. It does not feel good that nothing happens I think.

tpanajott, 2025-03-16:
I'm thinking that this may be easiest solved by simply restricting the users from deleting the last scenes & entities page from each room so that each room always has them. What do you think about that? It quickly becomes a mess in the backend if we want to somehow simulate pages that does not actually exist.

cablesandcoffee, 2025-03-16:
Yeah that sounds like a good idea! Maybe 4-page is the standard when a room
is created for example?

tpanajott, 2025-03-16:
Yeah, I'm thinking the same thing 👍

tpanajott, 2025-03-29:
Just to clarify, you have added them as capital X and Y and not x and y? Also, talking about icons, they are handled by the manager so they should probably be handled there instead of in the panel code.

tpanajott, 2025-03-29:
Answered my own question, it's lower case "x" and "y". Also, it looks so much better now. How do we wish to handle the icon colors? As in the image? Have the save icon highlighted to indicate that it can be saved and that then have the others inactive?

[image]

cablesandcoffee, 2025-03-30:
Oooh nice! Looking good. Yeah I think that's smart. Highlighted for NSPM scene. White for OH/HA scenes.
And yes my bad, it was lower case. ;)
About the icons. I didn't understand, should I add all the icons that we have in our custom font to the font mapping file that we already have? Currently I think there is only mappings for the weather icons?

tpanajott, 2025-03-30:
Then I will keep it as it is. It's now also implemented so that pressing on the text itself for the scene and not the icon will trigger it.

I have moved the icon_mapping from the icon_mappings.json file to a .hpp file (the same way we handle GUI_DATA in the firmware). This makes it much easier to handle and doesn't require the manager to interpret the file as it's just compiled into the code.

There is currently two files for icons, one is `docker/MQTTManager/include/weather/weather_icons.hpp` and the other one is `MQTTManager/include/entity/entity_icons.hpp` for icons related to the entities.

tpanajott, 2025-03-30:
So, scenes should now be fully working except for the "All lights"/"House" mode which is not currently implemented.

tpanajott, 2025-03-30:
Also, mixing NSPM and OpenHAB scenes causes issues as when activating an OpenHAB scene turning of a light issue #125 causes issues. It makes it so that when trying to activate an NSPM Scene it already thinks the light is on and so doesn't send the specific "Turn light on" command if the brightness to set it to is the same as it was previously.

cablesandcoffee, 2025-04-06:
Tested scenes now. So Openhab scenes works great. No problems from what I can see.

The NSPM scenes though does not work at all when I'm testing now? Nothing is sent to Openhab at all when trying to activate an NSPM scene.

This is the only thing I see in the logs:
'''[2025-04-06 09:36:47.375] [debug] [nspm_scene.cpp:95] [227] Saving scene 8::Scene 2.
[2025-04-06 09:36:47.376] [info] [nspm_scene.cpp:198] [227] Scene 8::Scene 2 saved successfully.'''

'''[2025-04-06 09:37:15.444] [debug] [entity_manager.cpp:504] [227] Will toggle entity in slot 3 in page with ID 81.
[2025-04-06 09:37:15.444] [info] [nspm_scene.cpp:51] [227] Activating scene 9::Scene 4.'''

Openhab log is empty which makes me think the manager is not sending any of the saved scene data to Openhab.

closed 2025-04-10 by cablesandcoffee