NSPManager/NSPanelManager #240: *DEVEL* ROOM PAGE - Can't use light switches or enter individual lights

cablesandcoffee, 2025-02-22:
**Describe the bug**
Light switches does not work. 
Can't enter individual lights pages.

tpanajott, 2025-02-22:
I belive that light switches should work but probably didn't because of the same problem as #235. Individual light pages aren't implemented yet.

cablesandcoffee, 2025-02-22:
Changed headline.

Switches now works to turn a light OFF but not ON. It sends 0 to the light when it's off and you try to turn it on. Same behavior on first page when pressing ceiling/table button that is off, it send 0 because the dimmer slider is at 0. I can open another issue for the first page all though I'm thinking they are connected.

tpanajott, 2025-02-25:
Hm, yes. There seems to be an issue with how it sends the toggle command so you can turn it off, but to then turn it on again you have to toggle it on, toggle it off and then the toggle on will work again.

Also, only the 4 entities page has been tested by me as I don't have the button IDs for the other pages. Perhaps you could double check my guesses in the GUI_data.hpp file over in the [firmware repo](https://github.com/NSPManager/NSPanelManager/blob/devel/firmware/firmware_espidf/lib/PageManager/GUI_data.hpp)?

cablesandcoffee, 2025-02-26:
aha I will check the button IDs!

cablesandcoffee, 2025-02-27:
Now I can use all the switches on the itemspages to turn lights off. I can never turn a light ON no matter how many times I press the switch.
I have one Item that is a Switch in Openhab. That can be turned ON and OFF as expected.

tpanajott, 2025-02-27:
Hm, that might be related to it not having a brightness as per the other bug. Try turning the light on.voa panel and then try toggling again.

cablesandcoffee, 2025-02-27:
Oh yeah that sounds reasonable. I think it sends brightness 0 when trying
to turn on so yes it should be solved with the other one. Good thinking! :)

tpanajott, 2025-03-16:
Hm, this should have been fixed together with issue #247 and if I set the manager log level to trace I see that the manager sends out a message to OpenHAB `{"payload":"{\"type\":\"Percent\",\"value\":60}","source":"NSPanelManager::MqttManager","topic":"openhab/items/OfficeCeilingLight_Brightness/command","type":"ItemCommandEvent"}` that to my knowledge about OpenHAB should turn the light on, right? It works for Home Assistant entities.

cablesandcoffee, 2025-03-28:
Nice! This works, great!

closed 2025-03-28 by cablesandcoffee