NSPManager/NSPanelManager #241: *DEVEL* Settings page - Error page when trying to save settings

cablesandcoffee, 2025-02-22:
**Describe the bug**
Can't save settings page. Results in error page.

`<h1>ZombieProcess
       at /save_settings</h1>
  <pre class="exception_value">PID still exists but it's a zombie (pid=447)</pre>
  
Request Method: | POST
-- | --
http://192.168.32.201:8010/save_settings
5.1.1
ZombieProcess
PID still exists but it's a zombie (pid=447)
/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py, line 1761, in _raise_if_zombie
web.views.save_settings
/usr/local/bin/uwsgi
3.12.5
['.',  '',  '/usr/local/lib/python312.zip',  '/usr/local/lib/python3.12',  '/usr/local/lib/python3.12/lib-dynload',  '/usr/local/lib/python3.12/site-packages']
Sat, 22 Feb 2025 08:08:38 +0000

`

tpanajott, 2025-02-22:
Can't replicate. Would need more logs unfortunately.

cablesandcoffee, 2025-02-22:
Hmm, I tried again now after pulling the latest from github. First time i press save it works. Seconds time i crashes and continues to do so. Also after restart of manager.

These are the logs from the manager after pressing save button and it fails:

Internal Server Error: /save_settings Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner response = get_response(request) ^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response response = wrapped_callback(request, *callback_args, **callback_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/app/nspanelmanager/web/views.py", line 626, in save_settings restart_mqtt_manager() File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline(): ^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline return self._proc.cmdline() ^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper return fun(self, *args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline self._raise_if_zombie() File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie raise ZombieProcess(self.pid, self._name, self._ppid) psutil.ZombieProcess: PID still exists but it's a zombie (pid=1806) Internal Server Error: /save_settings Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner response = get_response(request) ^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response response = wrapped_callback(request, *callback_args, **callback_kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/app/nspanelmanager/web/views.py", line 626, in save_settings restart_mqtt_manager() File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline(): ^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline return self._proc.cmdline() ^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper return fun(self, *args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline self._raise_if_zombie() File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie raise ZombieProcess(self.pid, self._name, self._ppid) psutil.ZombieProcess: PID still exists but it's a zombie (pid=1806) [pid: 313|app: 0|req: 164/728] 192.168.32.21 () {52 vars in 995 bytes} [Sat Feb 22 20:46:27 2025] POST /save_settings => generated 81148 bytes in 23 msecs (HTTP/1.1 500) 7 headers in 248 bytes (1 switches on core 0)

tpanajott, 2025-03-08:
This should be fixed, please retry.

cablesandcoffee, 2025-03-08:
yes fixed closing! 💯

closed 2025-03-08 by cablesandcoffee