NSPManager/NSPanelManager #242: *DEVEL* MANAGER ROOM SETTINGS - Can't add light/switch or edit entities

cablesandcoffee, 2025-02-22:
**Describe the bug**
Adding light/switch or edit entity throws error.

Log when pressing edit on light entity:

`Failed to get OpenHAB items!
Traceback (most recent call last):
  File "/usr/src/app/nspanelmanager/web/openhab_api.py", line 41, in get_all_openhab_items
    "label": item["label"],
             ~~~~^^^^^^^^^
KeyError: 'label'
Internal Server Error: /htmx/partial/edit_light_entity_in_room/11
Internal Server Error: /htmx/partial/edit_light_entity_in_room/11
[pid: 198|app: 0|req: 400/1246] 192.168.32.21 () {48 vars in 876 bytes} [Sat Feb 22 08:24:12 2025] GET /htmx/partial/edit_light_entity_in_room/11 => generated 94 bytes in 36 msecs (HTTP/1.1 500) 7 headers in 237 bytes (1 switches on core 0)
`

LOG WHEN ADDING LIGHT/SWITCH

`Internal Server Error: /htmx/partial/select_new_entity_source/ADD_LIGHT_TO_ROOM/{"page_id": 6, "page_slot": 0, "room_id": 3}
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/views/decorators/csrf.py", line 65, in _view_wrapper
    return view_func(request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/htmx.py", line 627, in partial_add_entity_to_entities_page_select_entity_source
    return redirect('htmx_partial_select_new_entity_config_modal', entity_source="openhab")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/shortcuts.py", line 49, in redirect
    return redirect_class(resolve_url(to, *args, **kwargs))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/shortcuts.py", line 180, in resolve_url
    return reverse(to, args=args, kwargs=kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/urls/base.py", line 88, in reverse
    return resolver._reverse_with_prefix(view, prefix, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/urls/resolvers.py", line 831, in _reverse_with_prefix
    raise NoReverseMatch(msg)
django.urls.exceptions.NoReverseMatch: Reverse for 'htmx_partial_select_new_entity_config_modal' not found. 'htmx_partial_select_new_entity_config_modal' is not a valid view function or pattern name.
Internal Server Error: /htmx/partial/select_new_entity_source/ADD_LIGHT_TO_ROOM/{"page_id": 6, "page_slot": 0, "room_id": 3}
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/views/decorators/csrf.py", line 65, in _view_wrapper
    return view_func(request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/htmx.py", line 627, in partial_add_entity_to_entities_page_select_entity_source
    return redirect('htmx_partial_select_new_entity_config_modal', entity_source="openhab")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/shortcuts.py", line 49, in redirect
    return redirect_class(resolve_url(to, *args, **kwargs))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/shortcuts.py", line 180, in resolve_url
    return reverse(to, args=args, kwargs=kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/urls/base.py", line 88, in reverse
    return resolver._reverse_with_prefix(view, prefix, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/urls/resolvers.py", line 831, in _reverse_with_prefix
    raise NoReverseMatch(msg)
django.urls.exceptions.NoReverseMatch: Reverse for 'htmx_partial_select_new_entity_config_modal' not found. 'htmx_partial_select_new_entity_config_modal' is not a valid view function or pattern name.
[pid: 194|app: 0|req: 135/1250] 192.168.32.21 () {48 vars in 1020 bytes} [Sat Feb 22 08:25:14 2025] GET /htmx/partial/select_new_entity_source/ADD_LIGHT_TO_ROOM/%7B%22page_id%22:%206,%20%22page_slot%22:%200,%20%22room_id%22:%203%7D => generated 83182 bytes in 12 msecs (HTTP/1.1 500) 7 headers in 248 bytes (1 switches on core 0)
`

tpanajott, 2025-03-08:
I believe  this could be fixed as of latest changes?

cablesandcoffee, 2025-03-08:
Yes fixed! :)

closed 2025-03-08 by cablesandcoffee