NSPManager/NSPanelManager #244: *DEVEL* - MANAGER ROOM PAGE - Delete room throws error and manager restart

cablesandcoffee, 2025-02-22:
**Describe the bug**
If you delete a room in the manager it seems like the manager restarts.

Log from manager when presseing delete room:

`PID still exists but it's a zombie (pid=2487)
PID still exists but it's a zombie (pid=2618)
PID still exists but it's a zombie (pid=2662)
PID still exists but it's a zombie (pid=2904)
Found running MQTTManager. Sending reload command via SIGUSR1 signal.
[2025-02-22 09:46:54.451] [info] [main.cpp:37] [3956] Reloading config from manager. Will wait 50ms for other processing to complete before proceeding with config reload.
[pid: 194|app: 0|req: 204/1760] 192.168.32.21 () {46 vars in 839 bytes} [Sat Feb 22 08:46:54 2025] GET /rooms/8/delete => generated 0 bytes in 18 msecs (HTTP/1.1 302) 7 headers in 232 bytes (1 switches on core 0)
{'ingress_path': '', 'theme': 'default', 'manager_address': '192.168.32.201', 'rooms': [{'room': <Room: Bedroom>, 'nspanels': <QuerySet []>}, {'room': <Room: Kitchen>, 'nspanels': <QuerySet []>}, {'room': <Room: Livingroom>, 'nspanels': <QuerySet [<NSPanel: nspaneltesterEU>, <NSPanel: US Panel>]>}, {'room': <Room: Bathroom>, 'nspanels': <QuerySet []>}, {'room': <Room: Entryway>, 'nspanels': <QuerySet []>}]}
`

tpanajott, 2025-02-25:
Hopefully this has been fixed as of latest commit. Please retry.

cablesandcoffee, 2025-02-26:
Working!

closed 2025-02-26 by cablesandcoffee