NSPManager/NSPanelManager #245: *DEVEL* MANAGER WEATHER PAGE - Save settings throws error

cablesandcoffee, 2025-02-22:
**Describe the bug**
Feels like the same issue as when saving on the settings page

[image]

cablesandcoffee, 2025-02-22:
These are the logs from the manager after pressing save button:

`Internal Server Error: /save_settings
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/views.py", line 626, in save_settings
    restart_mqtt_manager()
  File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager
    if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline():
                                                ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline
    return self._proc.cmdline()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper
    return fun(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline
    self._raise_if_zombie()
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie
    raise ZombieProcess(self.pid, self._name, self._ppid)
psutil.ZombieProcess: PID still exists but it's a zombie (pid=1806)
Internal Server Error: /save_settings
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/views.py", line 626, in save_settings
    restart_mqtt_manager()
  File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager
    if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline():
                                                ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline
    return self._proc.cmdline()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper
    return fun(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline
    self._raise_if_zombie()
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie
    raise ZombieProcess(self.pid, self._name, self._ppid)
psutil.ZombieProcess: PID still exists but it's a zombie (pid=1806)
[pid: 313|app: 0|req: 164/728] 192.168.32.21 () {52 vars in 995 bytes} [Sat Feb 22 20:46:27 2025] POST /save_settings => generated 81148 bytes in 23 msecs (HTTP/1.1 500) 7 headers in 248 bytes (1 switches on core 0)
`

cablesandcoffee, 2025-02-22:
same here as on settings page. Worked the first time I saved but not when trying again.

Log from error:

`Internal Server Error: /entities/weather_and_time
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/views.py", line 859, in weather_and_time
    restart_mqtt_manager()
  File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager
    if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline():
                                                ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline
    return self._proc.cmdline()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper
    return fun(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline
    self._raise_if_zombie()
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie
    raise ZombieProcess(self.pid, self._name, self._ppid)
psutil.ZombieProcess: PID still exists but it's a zombie (pid=5385, name='nspm_mqttmanage')
Internal Server Error: /entities/weather_and_time
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/views.py", line 859, in weather_and_time
    restart_mqtt_manager()
  File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager
    if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline():
                                                ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline
    return self._proc.cmdline()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper
    return fun(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline
    self._raise_if_zombie()
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie
    raise ZombieProcess(self.pid, self._name, self._ppid)
psutil.ZombieProcess: PID still exists but it's a zombie (pid=5385, name='nspm_mqttmanage')
[pid: 4522|app: 0|req: 147/435] 192.168.32.21 () {52 vars in 1036 bytes} [Sat Feb 22 22:03:27 2025] POST /entities/weather_and_time => generated 78577 bytes in 15 msecs (HTTP/1.1 500) 7 headers in 248 bytes (1 switches on core 0)`

tpanajott, 2025-03-08:
This is the same issue as #241 which should be fixed. Please retest and close if fixed.

cablesandcoffee, 2025-03-08:
Yes fixed! 👍

closed 2025-03-08 by cablesandcoffee