NSPManager/NSPanelManager #248: *DEVEL* MANAGER ROOMS PAGE - removing x button page make panel unresponsive

cablesandcoffee, 2025-02-22:
**Describe the bug**
Livingroom had 2 pages. When I removed the second one the panel became unresponsive.
Could not use any of the functions on first page on the panel. I can't reproduce this though but it has happened a couple of times. Need to figure out exactly what triggers it...

Log from panel when it became unresponsive:

`Time 	Level 	Message
10:52:25 PM	WARNING	(94872) event: handler already registered, overwriting
10:52:25 PM	ERROR	(94862) EntitiesPage: Unknown entities page type! Type: 0. Will navigate to home page.
10:52:21 PM	ERROR	(91472) ScreensaverPage: Trying to update screensaver page but not data to decode protobuf is available.
10:51:50 PM 	ERROR 	(59542) event: handler already registered, overwriting
10:51:50 PM 	ERROR 	(59542) EntitiesPage: Unknown entities page type! Type: 0. Will navigate to home page.
10:51:48 PM 	INFO 	(58142) NSPM_ConfigManager: Received new config data from MQTT.
10:51:48 PM 	INFO 	(58142) NSPM_ConfigManager: Received new config data from MQTT.
10:51:48 PM 	ERROR 	(58122) UpgradeManager: Failed to parse payload as JSON.
10:51:48 PM 	ERROR 	(58112) EntitiesPage: Unknown entities page type! Type: 0. Will navigate to home page.
10:51:48 PM 	DEBUG 	(58112) EntitiesPage: Received new data. Will update display.
10:51:48 PM 	DEBUG 	(58112) RoomManager: Received new entities page state update.`

tpanajott, 2025-03-15:
I think this should be fixed. This issue was that the manager sent to many updates for the panel to handle and as such it got stuck with a config with 0 rooms and it couldn't perform ANY actions with that.

cablesandcoffee, 2025-03-15:
There's still something here I think. 
If I do something like the following it sometimes triggers.
Sometimes it does not work to enter room page, sometimes its possible to enter room page but then the room page i s unresponsive.

1. I have an empty page called "test"
2. I add a page with 4 buttons
3. Panel works as normal
4. I delete the 4 button page so the test page is now empty
5. Panel is in screensaver mode
6. Press the panel to deactivate screensaver
7. I press the room button to open the default room page
8. Nothing happens
9. Change room, works
10. Enter room, works
11. Room page unresponsive
12. "unlocks" when screensaver activates

tpanajott, 2025-03-16:
I'm not quite following, for step 1 you say that you have an empty page, do you mean an empty room? Also, in step 10 you say that entering the room works but that the page in unresponsive? And what "unlocks" in screensaver activates in step 12?

cablesandcoffee, 2025-03-16:
Yeah sorry, room with no pages.
Yes I end up in a situation where I get to the room page on the panel but it doesn't respond to presses, can't use the "go back"-button for example. The unresponsivnes unlocks when screensaver activates. Then I can touch the panel enter a room and everything works.

cablesandcoffee, 2025-04-09:
Closing this one since we cannot reproduce it.

closed 2025-04-09 by cablesandcoffee