NSPManager/NSPanelManager #251: *DEVEL* MANAGER PANEL SETTINGS - both (no weather) screensaver does not populate any data

cablesandcoffee, 2025-02-22:
**Describe the bug**
time, date, temp is not populated. anything to do with the tft file?

tpanajott, 2025-02-25:
Well, having a look in the GUI_data.hpp file maybe it's wrong? The `screensaver_minimal_page_name` set set to "screensaver2" but, for example, the label name for current time `label_screensaver_minmal_current_time` is set to `screensaver.curTime`, I'm guessing that should be `screensaver2.curTime`?

cablesandcoffee, 2025-02-26:
Yes there seem to be a lot of errors actually:

`[2025-02-26 21:04:23.476] [debug] [light.cpp:40] [3445] Light 1::Bed left base loaded, can dim: yes, can color temp: yes, can_rgb: yes. Controlled from main page? Yes. Entities page ID 1 in slot 0.
[2025-02-26 21:04:23.483] [error] [openhab_light.cpp:344] [3445] Light 1::Bed left got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedLeft_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.485] [error] [openhab_light.cpp:368] [3445] Light 1::Bed left got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedLeft_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.485] [debug] [openhab_light.cpp:67] [3445] Loaded light 1::Bed left.
[2025-02-26 21:04:23.485] [info] [entity_manager.cpp:159] [3445] Light 1::Bed left was found in database but not in config. Creating light.
[2025-02-26 21:04:23.486] [debug] [light.cpp:50] [3445] Loading light 2::Bed Right.
[2025-02-26 21:04:23.486] [debug] [light.cpp:40] [3445] Light 2::Bed Right base loaded, can dim: yes, can color temp: yes, can_rgb: yes. Controlled from main page? Yes. Entities page ID 1 in slot 1.
[2025-02-26 21:04:23.490] [error] [openhab_light.cpp:344] [3445] Light 2::Bed Right got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedRight_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.492] [error] [openhab_light.cpp:368] [3445] Light 2::Bed Right got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item HayBedRight_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.492] [debug] [openhab_light.cpp:67] [3445] Loaded light 2::Bed Right.
[2025-02-26 21:04:23.492] [info] [entity_manager.cpp:159] [3445] Light 2::Bed Right was found in database but not in config. Creating light.
[2025-02-26 21:04:23.493] [debug] [light.cpp:50] [3445] Loading light 3::Desktop.
[2025-02-26 21:04:23.493] [debug] [light.cpp:40] [3445] Light 3::Desktop base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 2.
[2025-02-26 21:04:23.497] [error] [openhab_light.cpp:344] [3445] Light 3::Desktop got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item DesktopBlue_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.497] [debug] [openhab_light.cpp:67] [3445] Loaded light 3::Desktop.
[2025-02-26 21:04:23.497] [info] [entity_manager.cpp:159] [3445] Light 3::Desktop was found in database but not in config. Creating light.
[2025-02-26 21:04:23.498] [debug] [light.cpp:50] [3445] Loading light 4::Ceiling.
[2025-02-26 21:04:23.498] [debug] [light.cpp:40] [3445] Light 4::Ceiling base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 3.
[2025-02-26 21:04:23.502] [error] [openhab_light.cpp:344] [3445] Light 4::Ceiling got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item RetroGrandma_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.502] [debug] [openhab_light.cpp:67] [3445] Loaded light 4::Ceiling.
[2025-02-26 21:04:23.502] [info] [entity_manager.cpp:159] [3445] Light 4::Ceiling was found in database but not in config. Creating light.
[2025-02-26 21:04:23.503] [debug] [light.cpp:50] [3445] Loading light 16::Mirror.
[2025-02-26 21:04:23.503] [debug] [light.cpp:40] [3445] Light 16::Mirror base loaded, can dim: no, can color temp: no, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 4.
[2025-02-26 21:04:23.505] [debug] [openhab_light.cpp:67] [3445] Loaded light 16::Mirror.
[2025-02-26 21:04:23.505] [info] [entity_manager.cpp:159] [3445] Light 16::Mirror was found in database but not in config. Creating light.
[2025-02-26 21:04:23.506] [debug] [light.cpp:50] [3445] Loading light 17::Window.
[2025-02-26 21:04:23.506] [debug] [light.cpp:40] [3445] Light 17::Window base loaded, can dim: yes, can color temp: yes, can_rgb: no. Controlled from main page? Yes. Entities page ID 1 in slot 5.
[2025-02-26 21:04:23.510] [error] [openhab_light.cpp:344] [3445] Light 17::Window got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item TheBlue_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.510] [debug] [openhab_light.cpp:67] [3445] Loaded light 17::Window.
[2025-02-26 21:04:23.510] [info] [entity_manager.cpp:159] [3445] Light 17::Window was found in database but not in config. Creating light.
[2025-02-26 21:04:23.511] [debug] [light.cpp:50] [3445] Loading light 5::Sink.
[2025-02-26 21:04:23.511] [debug] [light.cpp:40] [3445] Light 5::Sink base loaded, can dim: yes, can color temp: no, can_rgb: no. Controlled from main page? Yes. Entities page ID 2 in slot 0.
[2025-02-26 21:04:23.513] [debug] [openhab_light.cpp:67] [3445] Loaded light 5::Sink.
[2025-02-26 21:04:23.513] [info] [entity_manager.cpp:159] [3445] Light 5::Sink was found in database but not in config. Creating light.
[2025-02-26 21:04:23.514] [debug] [light.cpp:50] [3445] Loading light 6::Stove.
[... cut ...]
[2025-02-26 21:04:23.519] [debug] [light.cpp:40] [3445] Light 12::Ledstrip mirror base loaded, can dim: yes, can color temp: yes, can_rgb: yes. Controlled from main page? Yes. Entities page ID 4 in slot 0.
[2025-02-26 21:04:23.523] [error] [openhab_light.cpp:344] [3445] Light 12::Ledstrip mirror got OpenHAB data from OpenHAB via initial ItemStateFetched event for OpenHAB item LedstripBathroom_ColorTemperature though current color temperature state was not a number.
[2025-02-26 21:04:23.525] [error] [openhab_light.cpp:368] [3445] Light 12::Ledstrip mirror got OpenHAB data from OpenHAB via initial ItemStateFetched event fo`

cablesandcoffee, 2025-02-27:
Hmm yes that should be screensaver2. since that's the reference to the page where the device is. I'll check the .hpp file now...

cablesandcoffee, 2025-02-27:
Checked the GUI_data.hpp file. Confirmed it had the wrong page name as reference so added screensaver2. where is was missing.

cablesandcoffee, 2025-02-27:
Now populating data. Closing

closed 2025-02-27 by cablesandcoffee