NSPManager/NSPanelManager #255: *DEVEL* - Manager settings save return error zombieProcess

cablesandcoffee, 2025-03-15:
**Describe the bug**
*DEVEL* - Manager settings save return error zombieProcess

It does not happen always but if repeatedly changing some settings and saving it throws the error.

Manager log when saving:

'''Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/views.py", line 633, in save_settings
    restart_mqtt_manager()
  File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager
    if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline():
                                                ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline
    return self._proc.cmdline()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper
    return fun(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline
    self._raise_if_zombie()
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie
    raise ZombieProcess(self.pid, self._name, self._ppid)
psutil.ZombieProcess: PID still exists but it's a zombie (pid=2073)
Internal Server Error: /save_settings
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/views.py", line 633, in save_settings
    restart_mqtt_manager()
  File "/usr/src/app/nspanelmanager/web/views.py", line 24, in restart_mqtt_manager
    if "/MQTTManager/build/nspm_mqttmanager" in proc.cmdline():
                                                ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/__init__.py", line 724, in cmdline
    return self._proc.cmdline()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1717, in wrapper
    return fun(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1856, in cmdline
    self._raise_if_zombie()
  File "/usr/local/lib/python3.12/site-packages/psutil/_pslinux.py", line 1761, in _raise_if_zombie
    raise ZombieProcess(self.pid, self._name, self._ppid)
psutil.ZombieProcess: PID still exists but it's a zombie (pid=2073)
[pid: 719|app: 0|req: 60/168] 172.17.0.1 () {52 vars in 994 bytes} [Sat Mar 15 08:32:16 2025] POST /save_settings => generated 81819 bytes in 53 msecs (HTTP/1.1 500) 7 headers in 248 bytes (1 switches on core 0)
WebSocketServer::handleConnection() HTTP status: 400 error: Error reading HTTP request line
WebSocketServer::handleConnection() HTTP status: 400 error: Error reading HTTP request line
[2025-03-15 09:32:22.057] [error] [nspanel.cpp:535] [2119] Caught exception: [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - invalid literal; last read: '<U+0018>'
[2025-03-15 09:32:22.057] [error] [nspanel.cpp:536] [2119] Stacktrace: Dynamic exception type: nlohmann::json_abi_v3_11_3::detail::parse_error
'''

web browser error:

[image]

tpanajott, 2025-03-15:
Yup, managed to trigger it and this should now be solved.

cablesandcoffee, 2025-03-15:
Can't trigger it anymore, closing!

closed 2025-03-15 by cablesandcoffee