NSPManager/NSPanelManager #256: *DEVEL* - MANAGER - Reboot button on first page does not work

cablesandcoffee, 2025-03-15:
**Describe the bug**
Reboot button on first page does not work

Pressing reboot throws red pop up in web browser:
Error while processing error response from server.

Manager log:
'''Forbidden (CSRF token missing.): /
Forbidden (CSRF token missing.): /
[pid: 463|app: 0|req: 39/202] 172.17.0.1 () {58 vars in 984 bytes} [Sat Mar 15 22:31:09 2025] POST / => generated 2491 bytes in 1 msecs (HTTP/1.1 403) 6 headers in 221 bytes (1 switches on core 0)
'''

tpanajott, 2025-03-16:
Yeah, it seems to be an issue when updating the component that HTMX doesn't quite register the correct attributes. I'm thinking we probably have to implement a workaround javascript function for it.

tpanajott, 2025-03-16:
This should be fixed as of latest release.

cablesandcoffee, 2025-03-28:
Yes works, closing! 👍

closed 2025-03-28 by cablesandcoffee