NSPManager/NSPanelManager #259: *DEVEL* PAGE NUMBERS on room pages are not correct

cablesandcoffee, 2025-03-28:
**Describe the bug**
If i have two entities pages added to a room and I enter the room it says Livingroom 1/2 which is correct.
But if I go to the next page it says Livingroom 3/2. Should be 2/2

tpanajott, 2025-03-28:
This may be a residual bug that is left. Try to reorder the pages In the room and it should fix itself.

tpanajott, 2025-04-01:
The correct number of scene pages should now be shown. Please try to add/remove/rearrange the normal entities pages for a room and check if it fixed the number of the normal entities pages.

tpanajott, 2025-04-01:
> This may be a residual bug that is left. Try to reorder the pages In the room and it should fix itself.

I found and fixed a seperate bug that might have caused this as well. Please try and check if it works as expected now.

cablesandcoffee, 2025-04-02:
Nope, this is still weird. 

1. Add a new room called "test"
2. Add two new pages, one with 8 and one with 12
3. Check panel, room order is correct. Test 1, Test 2, Test 3
4. Remove first page with 4 buttons
5. Check panel, at this stage my panel was unresponsive, had to wait for screensaver to kick in before i could use the panel
Anyway check panel again, room order is wrong: Test 2/2, Test 3/2

These pages were number 2 and 3 but should now be 1 and 2. Second value 2 is correct  but the first value is wrong

6. remove page 1 which with 8 buttons
7. Check panel, it now says: 3/1

Again second value is correct since only one page is left.
FIrst value is still 3, guessing that value is somehow left since we created the page the first time. It was number three when it was created.

Lot of the times while I was doing this my panel got stuck so there's something else going on here as well that we need to figure out. I will try to get some logs when it happens.

tpanajott, 2025-04-02:
The page order was not recalculated when removing a page and thus, as the first number is based in page order, it was all messed up. This is now fixed, I did however managed to trigger the bug you wrote about that it isn't possible to back out of the page and need to wait for the screensaver to kick in. It seems the issue is that the manager somehow sends, or the panel somehow interpret the manager data, that page type is 0 and not 4, 8 or 12, and thus cant' interpret any presses as it doesn't know what IDs to compare to. I will have to dig deep into this.

cablesandcoffee, 2025-04-04:
Okay cool will try the fix. 

Ah nice that you managed to trigger that one. I was about to try and trigger it again now but then you already no where to start looking.

cablesandcoffee, 2025-04-04:
Tested this now. It's better but I think there is one thing left. 

* Add two pages to a room. 
* Drag page 2 to page 1 position
* When toggling through the rooms by using next button on panel you will first get to "Room 2/2" then "Room 1/2"

tpanajott, 2025-04-04:
Hm, thought that was fixed. Will have to take another look at it.

tpanajott, 2025-04-05:
The manager sorted pages before they had updated their config and was therefore always one change behind. This should be fixed now.

cablesandcoffee, 2025-04-06:
Nice, Can confirm it's looks good on both scene pages and item pages. Good work! 💯

closed 2025-04-06 by cablesandcoffee