NSPManager/NSPanelManager #262: *DEVEL* Openhab switch added to room does not work

cablesandcoffee, 2025-04-07:
**Describe the bug**
An added Openhab switch to a room does not work.

Openhab logs shows this when trying to use the switch on the panel:

`2025-04-07 22:40:05.490 [WARN ] [re.io.websocket.event.EventWebSocket] - Failed to process deserialized event '{"payload":"{\"type\":\"OnOff\",\"value\":\"ON\"}","source":"NSPanelManager::MqttManager","topic":"openhab/items//command","type":"ItemCommandEvent"}': Topic must follow the format {namespace}/{entityType}/{entity}/{action}.`

I can see // between items and command, is there supposed to be something there, the specific entity?

tpanajott, 2025-04-07:
Well of course it didn't, hadn't fully implemented it even though I though I had. I've now way of testing this be it should now be fixed according to how I believe it should work.

cablesandcoffee, 2025-04-08:
Almost working now...

I've added a switch to a room.
I can use the switch on the panel an commands are sent to openhab and the switch is changed. When using the panel the icon behaves a bit weird though, kind of flickering.

And when controlling the switch from Openhab I can see that the icon is changing but it is not changing correctly.
Press on in Openhab the switch goes to OFF on panel. OFF in Openhab switch goes to ON on panel.

closed 2025-04-08 by cablesandcoffee