NSPManager/NSPanelManager #266: *DEVEL* - Global scenes - Error when moving scene to new button position in manager

cablesandcoffee, 2025-04-12:
**Describe the bug**

https://github.com/user-attachments/assets/18809b59-4d83-49a7-a92e-20b647ac513d

Log from manager:
```
[2025-04-12 13:36:31.765] [info] [nspanel.cpp:556] [210] Failed to parse data. Will try as protobuf.
[pid: 220|app: 0|req: 2/2] 192.168.32.21 () {44 vars in 848 bytes} [Sat Apr 12 11:36:41 2025] GET /entities/global_scenes => generated 93322 bytes in 22 msecs (HTTP/1.1 200) 8 headers in 365 bytes (1 switches on core 0)
Found running MQTTManager. Sending reload command via SIGUSR1 signal.
[2025-04-12 13:36:44.043] [info] [main.cpp:37] [200] Reloading config from manager. Will wait 50ms for other processing to complete before proceeding with config reload.
Internal Server Error: /htmx/partial/move_entity
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/views/decorators/csrf.py", line 65, in _view_wrapper
    return view_func(request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/htmx.py", line 701, in partial_move_entity
    return entities_pages.get(request, view='edit_room', room_id=new_entity_in_slot.room.id, is_scenes_pages=new_entity_in_slot.entities_page.is_scenes_page, is_global_scenes_page=(new_entity_in_slot.entities_page.room == None))
                                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'id'
Internal Server Error: /htmx/partial/move_entity
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
               ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/views/decorators/csrf.py", line 65, in _view_wrapper
    return view_func(request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/nspanelmanager/web/htmx.py", line 701, in partial_move_entity
    return entities_pages.get(request, view='edit_room', room_id=new_entity_in_slot.room.id, is_scenes_pages=new_entity_in_slot.entities_page.is_scenes_page, is_global_scenes_page=(new_entity_in_slot.entities_page.room == None))
                                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'id'
[pid: 220|app: 0|req: 3/3] 192.168.32.21 () {54 vars in 1070 bytes} [Sat Apr 12 11:36:44 2025] POST /htmx/partial/move_entity => generated 62860 bytes in 29 msecs (HTTP/1.1 500) 7 headers in 248 bytes (1 switches on core 0)
```

tpanajott, 2025-04-12:
Nice reporting with video evidence :) Should be fixed with latest commit.

cablesandcoffee, 2025-04-12:
Haha yeah wanted to try to upload a video. :) Confirm that it works now! :)

closed 2025-04-12 by cablesandcoffee