NSPManager/NSPanelManager #273: *DEVEL* - Changing relay ON/OFF settings does not take affect until after reboot

cablesandcoffee, 2025-04-19:
Not really a bug just a thought on the relay behavior.

I think a newly flashed panel will have the relay settings to default to off.
If the user wants to have it default to ON the user will change that setting and press save.
Since all other changes take affect straight away I think the user would expect that to change as well.

Personally I think it would be better. :)

tpanajott, 2025-04-19:
Hm, I though that was implemented but the panel I have to test with does not have any relays attached. Will have to take another look.

cablesandcoffee, 2025-04-19:
Hmm it seems to work the first time you do it with a freshly started panel.
So I change to ON, press save, I hear the relays turn ON.
Change to OFF, press save, nothing happens. Have to reboot for it to change. 

Same way around if the settings are reverse than in the example.

cablesandcoffee, 2025-04-26:
Works nice!

closed 2025-04-26 by cablesandcoffee