NSPManager/NSPanelManager #276: Weather data does not clear after "day roll over"

cablesandcoffee, 2025-04-26:
**Describe the bug**
Weather data does not clear after "day roll over"

**To Reproduce**
Steps to reproduce the behavior:
A newly started panel shows five days of weather starting with todays day first. When the day changes the current days data does not get removed. After four days the current days data will be in like the fourth column. Rebooting the panel clears the old days and sets the current day first.

tpanajott, 2025-04-26:
Actually it only showed the first 5 items for forecast the panel got so even if the day that was shown changed forecast it didn't update. This should be fixed now.

cablesandcoffee, 2025-04-26:
Ah nice! That why its been raining on me!

closed 2025-04-26 by cablesandcoffee