NSPManager/NSPanelManager #281: *DEVEL* Panel freezes when trying to open room

cablesandcoffee, 2025-05-01:
**Describe the bug**
A lot of times the panel freezes straight when trying to enter a room. It feels like it mostly happen when the screensaver has just been turned off. So one touch to turn off screensaver and then straight after a touch to try and enter a room.

Every time it happens you can see this in the log for the panel:
`08:16:25	ERROR	(300672886) EntitiesPage: Got new entity page data event but couldn't got new entity page data!`

**Video**

https://github.com/user-attachments/assets/1d2f2640-3b82-420b-8a16-8c6deae7a8a6

cablesandcoffee, 2025-05-04:
Sorry but it did not fix the freezing.
I found a way to trigger it now.

If you just press continuously on the top left corner you get the behavior. This can happen just after one press as well.
Interesting is that I have one panel that i can spam on the left corner like forever but it never happens with that panel. 

Video:

tpanajott, 2025-05-04:
I think it has to do with the WiFi connectivity. It seems that it works better with those that have better WiFi connectivity.

cablesandcoffee, 2025-05-04:
aha interesting, it makes sense with the once I have issues with! :)

cablesandcoffee, 2025-05-23:
Fixed, so nice!

closed 2025-05-23 by cablesandcoffee