NSPManager/NSPanelManager #284: V2 Beta - HA Addon - Unable to access settings

jbeardon, 2025-05-19:
**Describe the bug**
Attempting to access settings page in the NSPanelManager UI, results in error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Ui Setting page.
2. Error - error occured while rendering component 'alert'

**Expected behavior**
Setting page should load correctly

**Screenshots**

[image]

**Environment info (please complete the following information):**
 - Home Assistant (HASSOS - Latest
 - Brave
 - NSPanelManager version 2.0.4-beta 
 - NSPanel Firmware version - Not relevant (not a panel firmware issue)

**Additional context**
Please note that updating from Beta 1.04 to version 2 does not appear to pull in MQTT settings from previous version. Panel firmware cannot be updated (there is a ref to this in the swedish channel). But the settings page is still errors out, so I could not investigate further.

jbeardon, 2025-05-19:
I've also attached the HA addon logs in case they are of use. They appear to basically mirror the front end error.

[a5d2b728_nspanelmanager-beta_2025-05-19T13-49-05.605Z.log](https://github.com/user-attachments/files/20297925/a5d2b728_nspanelmanager-beta_2025-05-19T13-49-05.605Z.log)

emmaly, 2025-05-19:
I receive the same error, same situation.  Using Chrome instead of Brave, but I have no expectation browser is involved in the issue.  Otherwise identical versions and error condition regarding Settings page error, MQTT configuration issue.

* NSPM: 2.0.4-beta
* HA Core: 2025.5.2
* HA Supervisor: 2025.05.1
* HA OS: 15.2
* Hardware: rpi5-64, 8GB RAM

emmaly, 2025-05-19:
As a test, I did a full uninstall of the addon and checked the "remove all addon information" checkmark, and then reinstalled the same version `2.0.4-beta`, with same effective result:

```
Received exception while trying to communicate with MQTTManager.
```

I had to start the addon twice, and it loaded the UI in first-run wizard mode, which is good.  But after proceeding through the setup, it still has the failure to open the Settings page and gives a "Home Assistant not connected." error in the top-right corner.

Browser console says:
```
07:46:06.534 base.js?refresh=1747665966:17 Document ready, attaching functions.
07:46:06.534 base.js?refresh=1747665966:20 Hooking into HTMX websocket
07:46:06.625 base.js?refresh=1747665966:61 Connected to MQTTManager.
```

Interesting part of addon log seems to be about websocket issue:
```
[2025-05-19 07:47:21.909] [debug] [websocket_server.cpp:157] [112] Broadcasting active warnings to all connected weboscket clients.
[2025-05-19 07:47:21.909] [debug] [websocket_server.cpp:163] [112] Sending warnings, found warning: Home Assistant not connected.
[2025-05-19 07:47:21.909] [error] [home_assistant_manager.cpp:116] [112] Failed to connect to Home Assistant websocket. Reason: Expecting status 101 (Switching Protocol), got 401 status connecting to ws://supervisor/api/websocket, HTTP Status line: HTTP/1.1 401 Unauthorized
```

jbeardon, 2025-05-19:
I am seeing the same errors as @emmaly  (indeed, I went through the same uninstall-reinstall process). 
There are corresponding websocket errors in the HA Supervisor log occurring multiple times per minute indicating no API token is being provided. I may be wrong but this may be 'fixable' once the settings page becomes available again.

`2025-05-19 14:16:30.235 WARNING (MainThread) [supervisor.api.middleware.security] No API token provided for /api/websocket`

tpanajott, 2025-05-19:
> I am seeing the same errors as @emmaly  (indeed, I went through the same uninstall-reinstall process).
> There are corresponding websocket errors in the HA Supervisor log occurring multiple times per [...]

That's strange as it should noad the API-token straight from the shell environment variable. Will have to take an extra look tonight.

tpanajott, 2025-05-19:
Beta vs 2.0.5 is currently building which should fix this issue.

tpanajott, 2025-05-20:
New version now uploaded.

jbeardon, 2025-05-20:
> New version now available.

Just updated - everything appears to be working.
Thanks again.

closed 2025-05-20 by jbeardon

tpanajott, 2025-05-20:
> > New version now available.
> 
> Just updated - everything appears to be working.
> Thanks again.

Thank you for your quick testing!