NSPManager/NSPanelManager #285: Duplicate/Erroneous Weather Settings Beta 2,0.5

jbeardon, 2025-05-20:
**Describe the bug**
There are some duplicated and/or erroneous unit settings (possbily just the front end labels) on the weather entity config page.

[image]

Wind speed - duplicate MPH
Precipitation - Farenheit unit - presumably inches?

tpanajott, 2025-05-20:
Thanks for the report. Top bad i get blind to these kinds of things during development but yes, i do belive them to simply be the label that is wrong.

jbeardon, 2025-05-20:
No problem, that was my thought as well.
It may also be worth introducing some error handling if the units have not been selected on the weather page. For instance, I set my HA outdoor temperature sensor and attempted to save (without setting the preferred units), which resulted in a backend error page.
While it was obvious from the error (to me) that a value was not present, that may not be the case for a lot of users.

tpanajott, 2025-05-20:
> No problem, that was my thought as well.
> It may also be worth introducing some error handling if the units have not been selected on the weather page. For instance, I set my HA outdoor temperature [...]

That's good. I though I had set them to required but alas not. Would be really happy to accept a PR if you want or I can take a look later today. It's in the weather_and_time.html file. I can also confirm that it's simply the labels that are wrong and the correct values are set from the duplicates.

jbeardon, 2025-05-20:
> That's good. I though I had set them to required but alas not. Would be really happy to accept a PR if you want or I can take a look later today. It's in the weather_and_time.html file. I can also [...]

I'll give it a go - I've reviewed the html and looks like it's within my capabilities - never done a PR before so we will see how it goes!

jbeardon, 2025-05-20:
>

@tpanajott I've just tried to create a PR using devel against beta but cannot do so because I am not a collaborator. Must admit I have no idea what I am doing with Github. Should I be creating my own fork?

tpanajott, 2025-05-20:
Hmm, yes. I do believe that you have to create your own fork (or simply edit it Github web interface) and then you should be able to open a PR from that. @sebastienvermeille seems to know a lot more about Github than I do so perhaps he knows why it doesn't want to.

It seems from this guide https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork that you can create a fork, do your changes, then open a PR to get the changes pushed back to this repo.

tpanajott, 2025-05-20:
This should be fixed as of next release the the PR merged.

tpanajott, 2025-05-20:
2.0.6 is now out with this issue resolved.

closed 2025-05-20 by tpanajott