NSPManager/NSPanelManager #286: 2.0.5-beta does not connect to MQTT Manager

nickmcgill, 2025-05-20:
I have upgraded to 2.0.5-beta but it appears to be unable to connect to the MQTT manager, although the nspanel messages are appearing in MQTT explorer.
At the top of the web UI, I have
"Connecting to MQTTManager." with a button to restart the manager (it does not appear to achieve anything.
The panel is in an unknown state and if I try and update the firmware, I get "Request error but no message specified."

I don't know what logs my be of use in tracing the problem.

Installation info:
Add-on for 
Home Assistant OS 
Core 2025.5.2 
Supervisor 2025.05.1
Operating System 15.2
Frontend 20250516.0

Any ideas?
Thanks

tpanajott, 2025-05-20:
That doesn't sound quite right. It seems the manager is failing to start then but not quite sure why. Could you send over to log for the add-on/container?

tpanajott, 2025-05-20:
Hmm, strange. It seems like the migrations didn't go through. Just a stupid question, did you try to restart the addon? If so, please restart it and send over the log for that.

Lusquinhos, 2025-05-20:
same here, and also "no such column: web_light.controlled_by_nspanel_main_page"

[a5d2b728_nspanelmanager-beta_2025-05-20T09-28-40.758Z.txt](https://github.com/user-attachments/files/20340431/a5d2b728_nspanelmanager-beta_2025-05-20T09-28-40.758Z.txt)

Lusquinhos, 2025-05-20:
more prints

[image]
[image]

tpanajott, 2025-05-20:
It seems the migrations were not applied which is really strange. Could you perhaps enter the control for the container (not quite sure how in HA) but when you get in, could you run the following commands and post the result:
* `cd nspanelmanager`
* `python manage.py showmigrations`

It seems (from this guide https://community.home-assistant.io/t/execute-in-home-assistant-container-context/415031/2) that you can setup SSH into the HA machine and then SSH into it. After that you should be able to run `docker ps` and find a container running NSPanelManager with a name (ID will work just as well). After that run `docker exec -it <name/id> /bin/bash` (for example `docker exec -it nspanelmanager /bin/bash`. Then you can run the commands above.

nickmcgill, 2025-05-20:
This was done from within the terminal & ssh addon in haos (For future reference)

[showmigrations.txt](https://github.com/user-attachments/files/20343494/showmigrations.txt)

nickmcgill, 2025-05-20:
> Hmm, strange. It seems like the migrations didn't go through. Just a stupid question, did you try to restart the addon? If so, please restart it and send over the log for that.

I had restarted it, stopped it, started it...

nickmcgill, 2025-05-20:
In the interest of completeness, here's the log of a restart:

[a5d2b728_nspanelmanager-beta_2025-05-20T10-23-43.900Z.log](https://github.com/user-attachments/files/20343630/a5d2b728_nspanelmanager-beta_2025-05-20T10-23-43.900Z.log)

ehoberg, 2025-05-20:
Might have something to do with the fact that 2 migrations are named the same:

[image]

Makes sense that in some scenario's this will break the upgrade process.

tpanajott, 2025-05-20:
Good find, that might be it. It should be shown be the "show migrations" (which it is, I can see that lots of migrations are missing). The question is as to why it fails. Perhaps @nickmcgill could run the following command and check the output `python manage.py migrate`. It should then show an error message as to why it fails.

ehoberg, 2025-05-20:
Looks like that is indeed the case:

Running migrations:
  Applying web.0044_remove_nspanel_heap_used_pct_and_more...Traceback (most recent call last):
  File "/usr/src/app/nspanelmanager/manage.py", line 22, in <module>

.....

  File "/usr/local/lib/python3.12/site-packages/django/db/migrations/migration.py", line 118, in apply
    operation.state_forwards(self.app_label, project_state)
  File "/usr/local/lib/python3.12/site-packages/django/db/migrations/operations/fields.py", line 169, in state_forwards
    state.remove_field(app_label, self.model_name_lower, self.name)
  File "/usr/local/lib/python3.12/site-packages/django/db/migrations/state.py", line 258, in remove_field
    old_field = model_state.fields.pop(name)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'heap_used_pct'

Since 0043_remove_nspanel_heap_used_pct_and_more.py was already executed, the table 'heap_used_pct' does not exist and therefore cannot be removed.

Just trying to help out a little with debugging, since @tpanajott your probably really busy right now

nickmcgill, 2025-05-20:
Let me know if there's anything I can do to help test a fix.

nickmcgill, 2025-05-20:
I've tried removing the file in the docker container but, as it's in the image, each time I restart the add-on my changes are undone. (It's only taken me 4 attempts to realise this!)

I _could_ try to rebuild the docker image but I think that's beyond my skill set as is incorporating it in HA.

Thanks for your efforts, I'll wait.

tpanajott, 2025-05-20:
> I've tried removing the file in the docker container but, as it's in the image, each time I restart the add-on my changes are undone. (It's only taken me 4 attempts to realise this!)
>
> I _could_ try [...]

Fair enough. Thanks for you help anyways. I'll probably build a 2.0.6 tonight and publish it tomorrow morning as the armhf build takes like 6-7 hours.

tpanajott, 2025-05-20:
It was easy enough to test and verify that it didn't break anything. I'm currently building the new release so it should be available later tonight, hopefully soon enough that I'll be able to make it available for you guys before I need to head to bed.

tpanajott, 2025-05-20:
> Thank you very much.
> 
> I'll test it as soon as I can and let you know.

2.0.6 is now available.

nickmcgill, 2025-05-20:
I'm waiting for HA to let me know - do you know if there is a way of forcing an update?

tpanajott, 2025-05-20:
> I'm waiting for HA to let me know - do you know if there is a way of forcing an update?

No, unfortunetly. A restart of HA usually do it though.

nickmcgill, 2025-05-20:
Sorry, bad news...Same error in the web gui, but trying "manage.py migrate" gives
`root@a5d2b728-nspanelmanager-beta:/usr/src/app/nspanelmanager# python manage.py migrate
/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py:98: RuntimeWarning: Accessing the database during app initialization is discouraged. To fix this warning, avoid executing queries in AppConfig.ready() or when your app modules are imported.
  warnings.warn(self.APPS_NOT_READY_WARNING_MSG, category=RuntimeWarning)
Failed to populate Home Assistant addon settings.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 105, in _execute
    return self.cursor.execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/sqlite3/base.py", line 354, in execute
    return super().execute(query, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: no such table: web_roomentitiespage

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/app/nspanelmanager/web/apps.py", line 132, in ready
    create_entity_pages_for_all_rooms()
  File "/usr/src/app/nspanelmanager/web/apps.py", line 73, in create_entity_pages_for_all_rooms
    if not existing_pages.exists():
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/query.py", line 1288, in exists
    return self.query.has_results(using=self.db)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/sql/query.py", line 660, in has_results
    return compiler.has_results()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/sql/compiler.py", line 1542, in has_results
    return bool(self.execute_sql(SINGLE))
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/sql/compiler.py", line 1574, in execute_sql
    cursor.execute(sql, params)
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 122, in execute
    return super().execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 79, in execute
    return self._execute_with_wrappers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 92, in _execute_with_wrappers
    return executor(sql, params, many, context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 100, in _execute
    with self.db.wrap_database_errors:
  File "/usr/local/lib/python3.12/site-packages/django/db/utils.py", line 91, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 105, in _execute
    return self.cursor.execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/sqlite3/base.py", line 354, in execute
    return super().execute(query, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
django.db.utils.OperationalError: no such table: web_roomentitiespage
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, sessions, web
Running migrations:
  Applying web.0044_remove_nspanel_heap_used_pct_and_more...Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 105, in _execute
    return self.cursor.execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/sqlite3/base.py", line 354, in execute
    return super().execute(query, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: no such column: ""heap_used_pct""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/app/nspanelmanager/manage.py", line 22, in <module>
    main()
[... 35 lines cut ...]
    cursor.execute(sql, params)
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 122, in execute
    return super().execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 79, in execute
    return self._execute_with_wrappers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 92, in _execute_with_wrappers
    return executor(sql, params, many, context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 100, in _execute
    with self.db.wrap_database_errors:
  File "/usr/local/lib/python3.12/site-packages/django/db/utils.py", line 91, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 105, in _execute
    return self.cursor.execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/sqlite3/base.py", line 354, in execute
    return super().execute(query, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
django.db.utils.OperationalError: no such column: ""heap_used_pct""
`

Do you want the container logs?
(Happy to leave this until tomorrow - tempus fugit)

nickmcgill, 2025-05-20:
Container log:

[a5d2b728_nspanelmanager-beta_2025-05-20T20-19-08.387Z.log](https://github.com/user-attachments/files/20354756/a5d2b728_nspanelmanager-beta_2025-05-20T20-19-08.387Z.log)

nickmcgill, 2025-05-20:
showmigrations gives

`root@a5d2b728-nspanelmanager-beta:/usr/src/app/nspanelmanager# python manage.py showmigrations
/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py:98: RuntimeWarning: Accessing the database during app initialization is discouraged. To fix this warning, avoid executing queries in AppConfig.ready() or when your app modules are imported.
  warnings.warn(self.APPS_NOT_READY_WARNING_MSG, category=RuntimeWarning)
Failed to populate Home Assistant addon settings.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 105, in _execute
    return self.cursor.execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/sqlite3/base.py", line 354, in execute
    return super().execute(query, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: no such table: web_roomentitiespage

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/app/nspanelmanager/web/apps.py", line 132, in ready
    create_entity_pages_for_all_rooms()
  File "/usr/src/app/nspanelmanager/web/apps.py", line 73, in create_entity_pages_for_all_rooms
    if not existing_pages.exists():
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/query.py", line 1288, in exists
    return self.query.has_results(using=self.db)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/sql/query.py", line 660, in has_results
    return compiler.has_results()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/sql/compiler.py", line 1542, in has_results
    return bool(self.execute_sql(SINGLE))
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/models/sql/compiler.py", line 1574, in execute_sql
    cursor.execute(sql, params)
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 122, in execute
    return super().execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 79, in execute
    return self._execute_with_wrappers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 92, in _execute_with_wrappers
    return executor(sql, params, many, context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 100, in _execute
    with self.db.wrap_database_errors:
  File "/usr/local/lib/python3.12/site-packages/django/db/utils.py", line 91, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/utils.py", line 105, in _execute
    return self.cursor.execute(sql, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/django/db/backends/sqlite3/base.py", line 354, in execute
    return super().execute(query, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
django.db.utils.OperationalError: no such table: web_roomentitiespage
admin
 [X] 0001_initial
 [X] 0002_logentry_remove_auto_add
 [X] 0003_logentry_add_action_flag_choices
auth
 [X] 0001_initial
 [X] 0002_alter_permission_name_max_length
 [X] 0003_alter_user_email_max_length
 [X] 0004_alter_user_username_opts
 [X] 0005_alter_user_last_login_null
 [X] 0006_require_contenttypes_0002
 [X] 0007_alter_validators_add_error_messages
 [X] 0008_alter_user_username_max_length
 [X] 0009_alter_user_last_name_max_length
 [X] 0010_alter_group_name_max_length
 [X] 0011_update_proxy_permissions
[... 37 lines cut ...]
 [X] 0031_lightstate_color_temperature
 [X] 0032_nspanel_temperature
 [X] 0033_nspanelsettings
 [X] 0034_nspanel_md5_data_file_nspanel_md5_firmware_and_more
 [X] 0035_remove_nspanel_last_seen
 [X] 0036_relaygroupbinding_relaygroup
 [X] 0037_remove_relaygroup_relay_group_bindings_and_more
 [X] 0038_nspanel_register_relay1_as_light_and_more
 [X] 0039_scene_scene_type
 [X] 0040_scene_backend_name
 [X] 0041_nspanel_denied
 [X] 0042_nspanel_accepted
 [X] 0043_alter_nspanel_accepted
 [ ] 0044_remove_nspanel_heap_used_pct_and_more
 [ ] 0045_roomentitiespage
 [ ] 0046_light_entities_page_alter_light_room_view_position
 [ ] 0047_alter_light_entities_page_and_more
 [ ] 0048_light_controlled_by_nspanel_main_page
 [ ] 0049_switch
 [ ] 0050_roomentitiespage_is_scenes_page
 [ ] 0051_remove_nspanel_ip_address
 [ ] 0052_roomentitiespage_is_global_scenes_page_and_more
 [ ] 0053_remove_nspanel_register_relay1_as_light_and_more
 [ ] 0054_merge_20250504_1018
`

tpanajott, 2025-05-20:
I think the original issue has been resolved and it's now moved on to another issue. I would appreciate of someone could try to edit the "apps.py" file and comment out the lines for creating initial data. After that it should be able to do the "manager.py migrate", if that works it should be able to restart the add-on and it should start working.

Unfortunately I can't really look more into this issue tonight as I need to get up tomorrow morning and not sleep all morning ^^

nickmcgill, 2025-05-20:
Thanks for that - sleep well

tpanajott, 2025-05-22:
Hey, v2.0.7-beta is now out and should *again* hopefully fix this issue.

nickmcgill, 2025-05-22:
Sorry, bad news I'm afraid (also apologies for the slow response, I've been away from the computer today.
I've attached files from HA addon log, migrate and show migrations.
I'm concerned that I might be the only one with this problem ... 

[a5d2b728_nspanelmanager-beta_2025-05-22T15-54-01.343Z.log](https://github.com/user-attachments/files/20396016/a5d2b728_nspanelmanager-beta_2025-05-22T15-54-01.343Z.log)

[207betashowmigrations.txt](https://github.com/user-attachments/files/20396027/207betashowmigrations.txt)

[207migrate.txt](https://github.com/user-attachments/files/20396033/207migrate.txt)

nickmcgill, 2025-05-22:
Good news/bad news:

Show migrations works! 

[207betashowmigrations.txt](https://github.com/user-attachments/files/20398079/207betashowmigrations.txt)

*But* 

[image]

[a5d2b728_nspanelmanager-beta_2025-05-22T17-54-02.754Z.log](https://github.com/user-attachments/files/20398097/a5d2b728_nspanelmanager-beta_2025-05-22T17-54-02.754Z.log)

Sorry for all this - I do appreciate the effort you're putting in.

nickmcgill, 2025-05-22:
I did a full restart of HA too.

tpanajott, 2025-05-22:
Yes, it should also have sent a command to the old topic which would have been `nspanel/<panel_name>/command` can you see that?

nickmcgill, 2025-05-22:
It looks like it thinks it's off-line though

tpanajott, 2025-05-22:
Well, the status report says it's currently updating. Are you sure it isn't updating? It won't show on the manager unfortunately.

tpanajott, 2025-05-22:
It should show on the display of the panel itself with what it's doing and it's progress.

nickmcgill, 2025-05-22:
Ok (2 sets of stairs later), it's  now go the NSPanel manager logo and the address 192.168.4.1 showing. Should I connect to it?

tpanajott, 2025-05-22:
Hm, that's strange but also good. It should have automatically connected to WiFi but, 192.168.4.1 is the IP used in the new firmware so it did indeed update the firmware.

nickmcgill, 2025-05-22:
Thanks for that - I had to log in to the hotspot, set up wifi & mqtt.
It looks like the firmware updated ok, NSPanelManager only wanted to update the tft which it's now doing.

Hopefully that's fixed it. 

Sorry to have caused you the extra work. I think a large gin is in order!

tpanajott, 2025-05-22:
> Thanks for that - I had to log in to the hotspot, set up wifi & mqtt.
> It looks like the firmware updated ok, NSPanelManager only wanted to update the tft which it's now doing.
> 
> Hopefully that's fixed it. 
> 
> Sorry to have caused you the extra work. I think a large gin is in order!
> 

Hehe, no worries. This was a bit of a hint but still fun. Still can't quite figure out why only a few people got this issue and for the majority of users it just worked. Must have been a race-condition or something like that.

Glad you got it running again!

nickmcgill, 2025-05-22:
All working, thanks for all your help.

ungil-73, 2025-05-23:
Hi, still don't work for me. After upgrade the webui works, but coudn't connect to MQTT server. Try to reconnect, refresh MQTT datas in the settings but save doesn't works.

<html>
<body>
<h1 style="padding: 0px; margin: 0px; font-weight: normal; color: rgb(0, 0, 0); font-family: sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">MultiValueDictKeyError at /save_settings</h1><pre class="exception_value" style="padding: 0px; margin: 10px 0px; font-size: 1.5rem; white-space: pre-wrap; word-break: break-word; font-family: sans-serif; color: rgb(87, 87, 87); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">'home_assistant_token'</pre>
Request Method: | POST
-- | --
http://192.168.0.190:8000/save_settings
5.1.1
MultiValueDictKeyError
'home_assistant_token'
/usr/local/lib/python3.12/site-packages/django/utils/datastructures.py, line 86, in __getitem__
web.views.save_settings
/usr/local/bin/uwsgi
3.12.5
['.',  '',  '/usr/local/lib/python312.zip',  '/usr/local/lib/python3.12',  '/usr/local/lib/python3.12/lib-dynload',  '/usr/local/lib/python3.12/site-packages']
Fri, 23 May 2025 05:40:27 +0000

</body>
</html>

tpanajott, 2025-05-23:
> Hi, still don't work for me. After upgrade the webui works, but coudn't connect to MQTT server. Try to reconnect, refresh MQTT datas in the settings but save doesn't works.
> # MultiValueDictKeyError at /save_settings
> 
> 'home_assistant_token'
> 
> Request Method: | POST -- | -- http://192.168.0.190:8000/save_settings 5.1.1 MultiValueDictKeyError 'home_assistant_token' /usr/local/lib/python3.12/site-packages/django/utils/datastructures.py, line 86, in __getitem__ web.views.save_settings /usr/local/bin/uwsgi 3.12.5 ['.', '', '/usr/local/lib/python312.zip', '/usr/local/lib/python3.12', '/usr/local/lib/python3.12/lib-dynload', '/usr/local/lib/python3.12/site-packages'] Fri, 23 May 2025 05:40:27 +0000

That's correct. I'm planning on releasing a new version tonight that will fix this (issue #290).

closed 2025-07-13 by tpanajott