NSPManager/NSPanelManager #290: General settings

mosphorus, 2025-05-22:
When I try to save general settings

**To Reproduce**
Steps to reproduce the behavior:
1. Go to menu
2. Click on settings
3. Make any change and click save
4. <header id="summary" style="padding: 10px 20px; margin: 0px; border-bottom: 1px solid rgb(221, 221, 221); background: rgb(255, 255, 204); color: rgb(0, 0, 0); font-family: sans-serif; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><h1 style="padding: 0px; margin: 0px; font-weight: normal;">MultiValueDictKeyError at /save_settings</h1><pre class="exception_value" style="padding: 0px; margin: 10px 0px; font-size: 1.5rem; white-space: pre-wrap; word-break: break-word; font-family: sans-serif; color: rgb(87, 87, 87);">'home_assistant_token'</pre>
Request Method: | POST
-- | --
http://homeassistant.local:8000/save_settings
5.1.1
MultiValueDictKeyError
'home_assistant_token'
/usr/local/lib/python3.12/site-packages/django/utils/datastructures.py, line 86, in __getitem__
web.views.save_settings
/usr/local/bin/uwsgi
3.12.5
['.',  '',  '/usr/local/lib/python312.zip',  '/usr/local/lib/python3.12',  '/usr/local/lib/python3.12/lib-dynload',  '/usr/local/lib/python3.12/site-packages']
Thu, 22 May 2025 18:56:21 +0000

</div></main><footer id="explanation" style="padding: 10px 20px; margin: 0px; border-bottom: 1px solid rgb(221, 221, 221); background: rgb(238, 238, 238); color: rgb(0, 0, 0); font-family: sans-serif; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><p style="padding: 0px; margin: 0px;">You’re seeing this error because you have<span> </span><code style="padding: 0px; margin: 0px; font-size: 13px; white-space: pre-wrap; word-break: break-word;">DEBUG = True</code><span> </span>in your Django settings file. Change that to<span> </span><code style="padding: 0px; margin: 0px; font-size: 13px; white-space: pre-wrap; word-break: break-word;">False</code>, and Django will display a standard page generated by the handler for this status code.</p></footer>MultiValueDictKeyError at /save_settings
'home_assistant_token'
Request Method:	POST
Request URL:	http://homeassistant.local:8000/save_settings
Django Version:	5.1.1
Exception Type:	MultiValueDictKeyError
Exception Value:	
'home_assistant_token'
Exception Location:	/usr/local/lib/python3.12/site-packages/django/utils/datastructures.py, line 86, in __getitem__
Raised during:	web.views.save_settings
Python Executable:	/usr/local/bin/uwsgi
Python Version:	3.12.5
Python Path:	
['.',
 '',
 '/usr/local/lib/python312.zip',
 '/usr/local/lib/python3.12',
 '/usr/local/lib/python3.12/lib-dynload',
 '/usr/local/lib/python3.12/site-packages']
Server time:	Thu, 22 May 2025 18:56:21 +0000
Traceback [Switch to copy-and-paste view](http://homeassistant.local:8000/save_settings#)
/usr/local/lib/python3.12/site-packages/django/utils/datastructures.py, line 84, in __getitem__
            list_ = super().__getitem__(key)
                        ^^^^^^^^^^^^^^^^^^^^^^^^ …
Local vars
During handling of the above exception ('home_assistant_token'), another exception occurred:
/usr/local/lib/python3.12/site-packages/django/core/handlers/exception.py, line 55, in inner
                response = get_response(request)
                               ^^^^^^^^^^^^^^^^^^^^^ …
Local vars
/usr/local/lib/python3.12/site-packages/django/core/handlers/base.py, line 197, in _get_response
                response = wrapped_callback(request, *callback_args, **callback_kwargs)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ …
Local vars
/usr/src/app/nspanelmanager/web/views.py, line 591, in save_settings
    elif request.POST["home_assistant_token"] != "":
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ …
Local vars
/usr/local/lib/python3.12/site-packages/django/utils/datastructures.py, line 86, in __getitem__
            raise MultiValueDictKeyError(key)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ …
Local vars
Request information
USER
AnonymousUser

GET
No GET data

[... 605 lines cut ...]
USE_X_FORWARDED_PORT	
False
WSGI_APPLICATION	
'nspanelmanager.wsgi.application'
X_FRAME_OPTIONS	
'SAMEORIGIN'
YEAR_MONTH_FORMAT	
'F Y'
You’re seeing this error because you have DEBUG = True in your Django settings file. Change that to False, and Django will display a standard page generated by the handler for this status code.

**Expected behavior**
Get confirmation that settings has been saved

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment info (please complete the following information):**
 - Home assistant 
 - Browser Opera
 - NSPanelManager version 2.07
 - NSPanel Firmware version 2.0

**Additional context**
Panel is often not responsive and I can see at the time, at mqtt:
2025-05-22 20:12:49: Client NSPMPanel-E4:65:B8:49:48:F4 disconnected due to malformed packet.

tpanajott, 2025-05-22:
Hmm, that's kind of strange as I don't seem to be getting that error. However, I can implement an easy workaround and it will be fixed in the next release.

mosphorus, 2025-05-22:
> Hmm, that's kind of strange as I don't seem to be getting that error. However, I can implement an easy workaround and it will be fixed in the next release.

So many different hardwares etc. Hard to please everyone. Will wait patiently as you already doing the great job!

tpanajott, 2025-05-24:
Hmm, strange. It seems that your DB has missing tables. Are you running it as an HA addon or as a standalone container?

ungil-73, 2025-05-24:
Running on a raspberry PI 👍 Installation method
Home Assistant OS Core 2025.5.3
Supervisor  2025.05.1
Operating System  15.2
Frontend  20250516.0

Mosquito broker as a home assistant add-on

mosphorus, 2025-05-24:
In my case all is sorted. Needed to update firmware on panel as was less
responsive. Don't know where to check what firmware panel is though...

tpanajott, 2025-05-24:
> Running on a raspberry PI 👍 Installation method Home Assistant OS Core 2025.5.3 Supervisor 2025.05.1 Operating System 15.2 Frontend 20250516.0
> 
> Mosquito broker as a home assistant add-on

Hmm, you haven't had it running ever right? The easiest would perhaps be to reinstall the addon and choose to remove all data when uninstalling the addon.

tpanajott, 2025-05-24:
> In my case all is sorted. Needed to update firmware on panel as was less
> responsive. Don't know where to check what firmware panel is though...
> […](#)

Nice! The firmware on the panel is visible on the top of the page on the NSPanel itself. Navigate to the IP of the panel or simply press the "Visit" in the cog-menu from the first page for the panel.

mosphorus, 2025-05-24:
> Nice! The firmware on the panel is visible on the top of the page on the NSPanel itself. Navigate to the IP of the panel or simply press the "Visit" in the cog-menu from the first page for the panel.

Haha, never been there lol. Thank you!

ungil-73, 2025-05-24:
> Hmm, you haven't had it running ever right? The easiest would perhaps be to reinstall the addon and choose to remove all data when uninstalling the addon.

Before 2.0 it worked fine, Try to remove and reinstall, thanks for your help

tpanajott, 2025-07-13:
This seems to be working as of now. Closing.

closed 2025-07-13 by tpanajott